micro-ROS/micro_ros_platformio #4: [Windows10] Failed to build microros library due to invalid directory

AntumArk, 2022-04-08:
- Hardware description: ESP32, Windows10
- Version or commit hash: 87baf5eee14151e4c6debc2f4c78a222e2865a89

#### Steps to reproduce the issue

1. clone 4ef4b1868d7126e404480b1dc17bb16e1b12cfe7
2. open examples/ESP32 project with Platformio
3. try building

OR

1. Create new ESP32 platformio project
2. Add lines below to platformio.ini
```
microros_transport = serial
microros_distro = galactic
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio 
```
3. Try building

#### Expected behavior
code compiles with microros library

#### Actual behavior
```
Warning! Ignore unknown configuration option `microros_transport` in section [env:esp32dev]
Warning! Ignore unknown configuration option `microros_distro` in section [env:esp32dev]
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
All required Python pip packages are installed
Configuring esp32dev with transport serial
micro-ROS dev already downloaded
Building micro-ROS dev dependencies
The system cannot find the path specified.
micro-ROS already downloaded
Building micro-ROS library
'.' is not recognized as an internal or external command,
operable program or batch file.
NotADirectoryError: [WinError 267] The directory name is invalid: 'C:\\Users\\Zinc\\HomeProjects\\micro_ros_platformio\\examples\\ESP32\\.pio\\libdeps\\esp32dev\\micro_ros_platformio/build/aux':
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Zinc\.platformio\platforms\espressif32\builder\main.py", line 233:
    target_elf = env.BuildProgram()
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 922:
    if _is_builtin(spec):
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 914:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1035:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 61:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 149:
    self.process_extra_options()
  File "C:\Users\Zinc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 292:
    self.env.SConscript(
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Zinc\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Zinc\HomeProjects\micro_ros_platformio\examples\ESP32\.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 74:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "C:\Users\Zinc\HomeProjects\micro_ros_platformio\examples\ESP32\.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 153:
    self.package_mcu_library()
  File "C:\Users\Zinc\HomeProjects\micro_ros_platformio\examples\ESP32\.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 220:
    os.makedirs(self.build_folder + "/aux", exist_ok=True)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3312.0_x64__qbz5n2kfra8p0\lib\os.py", line 225:
    mkdir(name, mode)
```

#### Additional information

AntumArk, 2022-04-08:
Tried debugging extra_script.py, but I get Import("env") that it is undefined. Same in Debian too.

AntumArk, 2022-04-08:
The suspected line is https://github.com/micro-ROS/micro_ros_platformio/blob/87baf5eee14151e4c6debc2f4c78a222e2865a89/microros_utils/library_builder.py#L207

./install/setup.sh is not present

AntumArk, 2022-04-08:
Replacing that line with
`os.system("bash && cd {} && . {}/install/setup.sh && {} > /dev/null 2>&1".format(self.mcu_folder, self.dev_folder, colcon_command))`
Allowed further execution. and found out a really odd thing.. Windows cant have a folder called "aux" ... Tried this also on windows explorer.

AntumArk, 2022-04-08:
Replaced all aux folders to aux1. Now the next error:
https://github.com/micro-ROS/micro_ros_platformio/blob/87baf5eee14151e4c6debc2f4c78a222e2865a89/microros_utils/library_builder.py#L232
cant find this file

thijstriemstra, 2022-04-22:
I'm also running into this issue. After removing the .pio directory, running install again, it fails with this error (using a powershell window in platformio):

```
C:\Users\Thijs\.platformio\penv\Scripts\python.exe -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in c:\users\thijs\.platformio\penv\lib\site-packages (2.0.1)
WARNING: You are using pip version 21.3.1; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\Users\Thijs\.platformio\penv\Scripts\python.exe -m pip install --upgrade pip' command.
Configuring az-delivery-devkit-v4 with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Build dev micro-ROS environment failed: 
 'colcon' is not recognized as an internal or external command,

operable program or batch file.
```

I guess I'll switch back to microros for arduino instead.

Acuadros95, 2022-08-23:
Closing, support for windows is not currently under development due to `colcon` cross compilation issues on windows.

Will update if progress is made.

closed 2022-08-23 by Acuadros95