micro-ROS/micro_ros_platformio #13: Running `cleanall` target takes a long time to finish

bjsowa, 2022-04-20:
I was trying to rebuild the library after changing some settings. Running:
```
pio run
```
again did not rebuild the library so I tried to clean build artifacts first by running:
```
pio run -t clean
```
This also didn't work because the build artifacts of micro-ROS library are not recognized as project's build artifacts by PlatformIO.
I then tried the `cleanall` target:
```
pio run -t cleanall
```
which should completely remove the cloned library and its artifacts. It worked but I noticed it takes a really long time to finish, almost 10 minutes in my case. 

I found out that running:
```
rm -rf .pio/libdeps
```
works but it doesn't feel "proper" for me.

Acuadros95, 2022-04-20:
We are currently implementing a clean target for microros, right now you can trigger a library rebuild deleting the `.pio/libdeps/<...>/micro_ros_platformio/build` folder.

Will give and update on this as soon as its implemented

Acuadros95, 2022-04-21:
Closing, reopen if problems appear with the added target.

> which should completely remove the cloned library and its artifacts. It worked but I noticed it takes a really long time to finish, almost 10 minutes in my case.

This should also be fixed.