micro-ROS/micro_ros_platformio #20: Compilation Error

v-mifass, 2022-04-26:
## Issue template

- Hardware description: Teensy 4.1
- RTOS: micro-ROS arduino

- Version or commit hash: main

#### Steps to reproduce the issue
platform.ini file:

[env:teensy41]
platform = teensy
board = teensy41
board_microros_distro = galactic
board_microros_transport = serial
framework = arduino
lib_deps = 
	https://github.com/micro-ROS/micro_ros_platformio

#### Expected behavior
compile

#### Actual behavior
Error

Requirement already satisfied: markupsafe==2.0.1 in c:\users\v-mifass\.platformio\penv\lib\site-packages (2.0.1)
Configuring teensy41 with transport serial
Downloading micro-ROS dev dependencies
ament_cmake clone failed: 
fatal: Too many arguments.

#### Additional information

Acuadros95, 2022-04-27:
Could not replicate this. What is your git version? 
`git --version`

pablogs9, 2022-04-27:
Should we check if `git` executable is in the path before running it? @Acuadros95

pablogs9, 2022-04-27:
Also, because of the `c:\` I guess that @v-mifass is running Windows. Maybe we should explore the possibility of porting this to Windows.

v-mifass, 2022-04-27:
yes running platformio on windows.  Git version 2.35.1.windows.2

v-mifass, 2022-04-28:
So compiling in Linux does work.  This can be closed unless a version to compile on platformio on windows is an 'issue'

Acuadros95, 2022-08-23:
Closing, support for windows is not currently under development due to colcon cross compilation issues on windows.

closed 2022-08-23 by Acuadros95