micro-ROS/micro_ros_platformio #24: Lots of compilation errors when invoking "pio run" for the [env:esp32dev_wifi] pio environment

safdark, 2022-05-03:
## Issue template
Several compilation errors when compiling for the [env:esp32dev_wifi] environment.

- Hardware description: esp32
- RTOS: arduino
- Installation type: micro_ros_platformio
- Version or commit hash: master

#### Steps to reproduce the issue

- $ git clone ```git@github.com:safdark/micro_ros_platformio_ci.git``` ./ci
(This is a copy of the ci/ folder from the original repository, with only the esp32dev_wifi environment present in the platformio.ini file, for simplicity)
- $ cd ci

- $ pio lib install
```
Library Storage: /home/safdar/code/repositories/myproject_firmware/entrypoints/ci/.pio/libdeps/esp32dev_wifi
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio
git version 2.25.1
Cloning into '/home/safdar/.platformio/.cache/tmp/pkg-installing-fvshhh20'...
remote: Enumerating objects: 64, done.
remote: Counting objects: 100% (64/64), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 64 (delta 10), reused 42 (delta 5), pack-reused 0
Unpacking objects: 100% (64/64), 123.23 KiB | 882.00 KiB/s, done.
Library Manager: micro_ros_platformio @ 0.0.1+sha.c20916a has been installed!

```

- $ pio run
#### Expected behavior
Should compile succesfully since I haven't changed anything in the original ci/ folder.

#### Actual behavior

Output:
```
$ pio run

Processing esp32dev_wifi (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/safdar/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/safdar/.platformio/penv/lib/python3.8/site-packages (6.0)
WARNING: You are using pip version 21.3.1; however, version 22.0.4 is available.
You should consider upgrading via the '/home/safdar/.platformio/penv/bin/python -m pip install --upgrade pip' command.
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/safdar/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/safdar/.platformio/penv/lib/python3.8/site-packages (2.0.1)
WARNING: You are using pip version 21.3.1; however, version 22.0.4 is available.
You should consider upgrading via the '/home/safdar/.platformio/penv/bin/python -m pip install --upgrade pip' command.
Configuring esp32dev with transport wifi
Downloading micro-ROS dev dependencies
	 - Downloaded ament_cmake
	 - Downloaded ament_lint
	 - Downloaded ament_package
	 - Downloaded googletest
	 - Downloaded ament_cmake_ros
	 - Downloaded ament_index
[... 387 lines cut ...]
colcon-package-selection 0.2.10: up-to-date
colcon-parallel-executor 0.2.4: up-to-date
colcon-pkg-config 0.1.0: up-to-date
colcon-powershell 0.3.7: up-to-date
colcon-python-setup-py 0.2.7: up-to-date
colcon-recursive-crawl 0.2.1: up-to-date
colcon-ros 0.3.22: newer version available (0.3.23)
colcon-test-result 0.3.8: up-to-date
colcon-zsh 0.4.0: up-to-date
```

Environment:
```
$ env | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=galactic
```

PIO version:
```
$ pio --version
PlatformIO Core, version 5.2.5
```

safdark, 2022-05-03:
@Acuadros95 , my previous output was:
```
$ env | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```
...because I read somewhere that ***only*** ```FastRTPS``` is guaranteed to work with MicroROS.

In any case, I changed it to ```RMW_IMPLEMENTATION=rmw_microxrcedds```, as per your recommendation, and did a clean build, but I still get the same errors as above.

Acuadros95, 2022-05-04:
I think you have ROS2 sourced by default on your terminal, at least its sourced on the environment you provided:
`AMENT_PREFIX_PATH=/opt/ros/galactic`

Please try with this PR, which removes ROS2 sourced references out of the micro-ROS build process: https://github.com/micro-ROS/micro_ros_platformio/pull/25

safdark, 2022-05-06:
Thanks @Acuadros95 !!
I incorporated that PR (i,e the ```fix/ros_source``` branch) and that worked!

safdark, 2022-05-06:
Btw, regarding the ```RMW_IMPLEMENTATION``` setting, as you mentioned, this package requires that that be set to ```rmw_microxrcedds```. However, I had read elsewhere that only ```rmw_fastrtps_cpp``` is guaranteed to work with MicroROS. Is that not a constraint any more?

Acuadros95, 2022-05-09:
Yes, only `rmw_fastrtps_cpp` is compatible on **ROS2 side**.

This means that you need to use FastRTPS as your ROS2 middleware when using ros2 console commands or running a `rclcpp` node, but micro-ROS uses `rmw_microxrcedds` internally. 

So when generating the micro-ROS binary, `rmw_microxrcedds` shall be used.

closed 2022-05-09 by Acuadros95

1larity, 2022-06-12:
> Thanks @Acuadros95 !! I incorporated that PR (i,e the `fix/ros_source` branch) and that worked!

Could someone explain how to do this, I have the same problem, but im a bit of a git noob.

Acuadros95, 2022-06-13:
@1larity Could you explain your problem? The fix for this issue is already on the latest main branch.