micro-ROS/micro_ros_platformio #28: set_microros_wifi_transports() invocation hangs with provided example

alikureishy, 2022-05-27:
## Issue template
- Hardware description: esp32dev
- RTOS: arduino
- Installation type: pio
- Version or commit hash: main

#### Steps to reproduce the issue
I'm trying to get the [publisher example](https://github.com/micro-ROS/micro_ros_platformio/tree/main/examples/micro-ros_publisher) working over ***wifi***, with the following platformio.ini file contents:
```
[env]
monitor_speed = 115200
upload_speed = 921600

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_microros_transport = wifi
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```

The code in [Main.cpp](https://github.com/micro-ROS/micro_ros_platformio/blob/main/examples/micro-ros_publisher/src/Main.cpp) is mostly unchanged, except for the wifi setup steps as instructed in the ```micro_ros_platformio``` [documentation](https://github.com/micro-ROS/micro_ros_platformio#transport-configuration):
```
const char* ssid = "My-Wifi-Network";
const char* password = "My-Wifi-Password";
const size_t agent_port = 8888;
IPAddress agent_ip(192, 168, 1, 219);
...

void setup() {
    Serial.begin(115200);
    Serial.print("Initializing...");
    set_microros_wifi_transports((char*)ssid, (char*)password, agent_ip, agent_port);
    Serial.print("Done!");
}
...
```

I'm also running the agent on the host using docker, as follows:
```
> docker run -it --rm --net=host microros/micro-ros-agent:galactic udp4 -p 8888 -v 6
```
...with output:
```
[1653609707.462781] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1653609707.463000] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```

I then ran the following steps to build and upload the uros client:
```
> pio lib install
    <Succeeded>
> pio run
    <Succeeded>
> pio run -t upload
   <Succeeded>
> pio device monitor
```

#### Expected behavior

I should start seeing messages being published to a ROS2 topic.

#### Actual behavior

```set_microros_wifi_transports()``` hangs, and there is nothing printed to the serial monitor after that point:

```
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Initializing...
```

- As per the code, a ```Done!``` should be printed after the transport initialization. However, it never gets to that point.
- Also, on the agent, no additional logs show up beyond the two lines that were printed at agent startup:
```
> docker run -it --rm --net=host microros/micro-ros-agent:galactic udp4 -p 8888 -v 6
[1653609707.462781] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1653609707.463000] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```

What am I missing?

#### Additional information

The host computer (on which the agent is running) has the following ifconfig output (which I have reflected in the wifi connection params in Main.cpp):
```
wlp4s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.219  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::65cb:1978:7797:6d81  prefixlen 64  scopeid 0x20<link>
        ether f8:63:3f:df:fe:ad  txqueuelen 1000  (Ethernet)
        RX packets 1865460  bytes 1437880778 (1.4 GB)
        RX errors 0  dropped 6  overruns 0  frame 0
        TX packets 530456  bytes 108092270 (108.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

I've separately verified that the ESP32 board's wifi IS working (using the Wifi.h library):
```
Connecting to wifi........
WiFi connected!
Assigned IP Address: 192.168.1.44
```

My environment settings on the agent/host are:
```
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=galactic
```

Acuadros95, 2022-05-30:
Please try with this fix: https://github.com/micro-ROS/micro_ros_platformio/issues/28

Acuadros95, 2022-06-01:
Closing, reopen if the problem persist after the fix

closed 2022-06-01 by Acuadros95