micro-ROS/micro_ros_platformio #30: Problem Building under Windows 10

quaiky01, 2022-05-30:
## Issue template

- Hardware description: Arduino zero
- Version or commit hash: Galaxy

#### Steps to reproduce the issue
building works under linux, but build on windows fails during build of micro-ROS dev environment (gtest_vendor)

#### Expected behavior

#### Actual behavior

#### Additional information
Error message:
 --- stderr: gtest_vendor

Traceback (most recent call last):

  File "C:\Users\isk5wi\.platformio\penv\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__

    rc = await self.task(*args, **kwargs)

  File "C:\Users\isk5wi\.platformio\penv\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__

    return await task_method(*args, **kwargs)

  File "C:\Users\isk5wi\.platformio\penv\lib\site-packages\colcon_ros\task\cmake\build.py", line 35, in build

    rc = await extension.build(

  File "C:\Users\isk5wi\.platformio\penv\lib\site-packages\colcon_cmake\task\cmake\build.py", line 87, in build

    rc = await self._reconfigure(args, env)

  File "C:\Users\isk5wi\.platformio\penv\lib\site-packages\colcon_cmake\task\cmake\build.py", line 153, in _reconfigure

    raise RuntimeError(

RuntimeError: VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.

---

Failed   <<< gtest_vendor [2.80s, exited with code 1]

VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.

huigang39, 2022-06-18:
# Same problem

Build dev micro-ROS environment failed: 
 --- stderr: gtest_vendor

Traceback (most recent call last):

  File "C:\Users\huigang\.platformio\penv\lib\site-packages\colcon_core\executor\__init__.py", line 

Failed   <<< gtest_vendor [1.26s, exited with code 1]

**VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.**

=================================== [FAILED] Took 128.55 seconds ===================================

huigang39, 2022-06-18:
> # Same problem
> Build dev micro-ROS environment failed: --- stderr: gtest_vendor
>
> Traceback (most recent call last):
>
> File [...]

I don't have **Visual Studio** installed, does that mean I need to install it for the build to succeed?

Acuadros95, 2022-08-23:
Closing, support for windows is not currently under development due to `colcon` cross compilation issues on windows.

Will update if progress is made.

> I don't have Visual Studio installed, does that mean I need to install it for the build to succeed?

Yes, colcon needs a Visual Studio environment. 
This is one of the main issues with porting the library cross compilation to Windows.

closed 2022-08-23 by Acuadros95