micro-ROS/micro_ros_platformio #32: Intellisense in VS Code not working with micro_ros_platformio

DaniGarciaLopez, 2022-06-02:
First of all I want to thank you guys to support PlatformIO, we really appreciate it! 

Previously we were working in PlatformIO with micro_ros_arduino. I ported the project to this new repo and everything compiles fine except that Intellisense in VS Code is not recognizing `#include <micro_ros_platformio.h>` anymore, so syntax highlight in micro-ROS is not working.

PlatformIO correctly adds `micro_ros_platformio` in `c_cpp_properties.json`, I also tried to rebuild Intellisense index with no luck. Any help is welcome. Thanks!

- Hardware description: Teensy 3.2
- Version or commit hash: Ubuntu 20.04 galactic, main branch last commit, platformio 6.0.1, vs code 1.67.2, c/c++ extension 1.9.8

#### Steps to reproduce the issue
- Clone this repository
- Go to `/ci` folder
- Open VS Code
- Check that PlatformIO and C/C++ extensions are installed and active
- Open PlatformIO terminal:
```
pio lib install
pio run
pio run --target upload
```
#### Expected behavior
Intellisense should work, `#include <micro_ros_platformio.h>` is found and code highlight works

#### Actual behavior
Intellisense is not working, `#include <micro_ros_platformio.h>` not found. 
Error: `#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit`

Genozen, 2022-07-18:
I have similar issue, but mine builds regardless...
Have you had any success getting the board running?

DaniGarciaLopez, 2022-07-18:
Hi @Genozen. Mine also builds fine, just Intellisense is not working. I haven't found a solution, if you do, please let me know!

Acuadros95, 2022-07-28:
Should be fixed with https://github.com/micro-ROS/micro_ros_platformio/pull/44. Please check it out!

closed 2022-07-28 by Acuadros95

DaniGarciaLopez, 2022-07-28:
I've just tested and it works fine! Thanks @bjsowa @Acuadros95

PR #44 (2022-07-28): Fix generation of idedata
Fixes #32 .
I wrote this PR only to demonstrate the problem, not to provide the best solution.
AFAIK, running `pio init --ide <ide name>` runs the `idedata` target to generate `idedata.json` file which contains information about include paths, defines etc. which is used to generate IDE-specific configuration files. 
Currently the `extra_script.py` completely ignores the `idedata` target. This was done to avoid building the library when IDE fetches project metadata.
I think the script should instead update some of the environments (probably the `global_env` and `projen [... cut]