micro-ROS/micro_ros_platformio #33: Initializing parameter server fails

fischejo, 2022-06-12:
## Issue template

Thank you for supporting platformio! I have problems to use the parameter-feature on galactic and with micro_ros_platformio. Therefore I reproduced the issue with this [example](https://github.com/micro-ROS/micro-ROS-demos/blob/galactic/rclc/parameter_server/main.c). 
The command `rclc_parameter_server_init_default` fails like the resources are not configured well. Did someone stumble across the same problem?

- Hardware description: Teensy 4.1
- Version or commit hash: ROS2 Galactic
  - repo: micro_ros_platformio
    - branch: main 
    - commit: 285dd946de9c53f1e0de4f3fc0033a1736d36062
  - repo: micro_ros_setup: 
    - branch: galactic 
    - tag: 2.1.0 
    - commit: aed3ad405c1a68ad6374ee1247bee68f87d7c65e
    - agent compiled as described in [teensy_with_arduino](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/)

#### Steps to reproduce the issue
1. Install agent
2. Setup platformio project

<details><summary>src/main.ino</summary>
<p>

```c
#include <micro_ros_platformio.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rclc_parameter/rclc_parameter.h>

#include <std_msgs/msg/int32.h>

#include <stdio.h>
#include <unistd.h>

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){Log.printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc); return 1;}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){Log.printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

#define Log Serial2

rcl_publisher_t publisher;
rclc_parameter_server_t param_server;
rcl_timer_t timer;
bool publish = true;
std_msgs__msg__Int32 msg;

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
	(void) last_call_time;
	(void) timer;

	if (publish)
	{
		RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
		Log.printf("Sent: %d\n", msg.data);

		msg.data++;
	}
}

void on_parameter_changed(Parameter * param)
{
	if (strcmp(param->name.data, "publish_toogle") == 0 && param->value.type == RCLC_PARAMETER_BOOL)
	{
		publish = param->value.bool_value;
		Log.printf("Publish %s\n", (publish) ? "ON" : "OFF");
	}
	else if (strcmp(param->name.data, "publish_rate_ms") == 0 && param->value.type == RCLC_PARAMETER_INT)
	{
		int64_t old;
		RCSOFTCHECK(rcl_timer_exchange_period(&timer, RCL_MS_TO_NS(param->value.integer_value), &old));
		Log.printf("Publish rate %ld ms\n", param->value.integer_value);
	}
}

int main()
{
    Log.begin(9600);

    Serial.begin(115200);
    set_microros_serial_transports(Serial);

	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "micro_ros_node", "", &support));

	// create publisher
	RCCHECK(rclc_publisher_init_default(
		&publisher,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
		"micro_ros_pub"));

  	// Create parameter service
    RCCHECK(rclc_parameter_server_init_default(&param_server, &node));

	// create timer,
	RCCHECK(rclc_timer_init_default(
		&timer,
		&support,
		RCL_MS_TO_NS(1000),
		timer_callback));

	// create executor
	rclc_executor_t executor = rclc_executor_get_zero_initialized_executor();
	RCCHECK(rclc_executor_init(&executor, &support.context, RCLC_PARAMETER_EXECUTOR_HANDLES_NUMBER + 1, &allocator));
	RCCHECK(rclc_executor_add_parameter_server(&executor, &param_server, on_parameter_changed));
	RCCHECK(rclc_executor_add_timer(&executor, &timer));

	// Add parameters
    RCCHECK(rclc_add_parameter(&param_server, "publish_toogle", RCLC_PARAMETER_BOOL));
    RCCHECK(rclc_add_parameter(&param_server, "publish_rate_ms", RCLC_PARAMETER_INT));

    //rclc_parameter_set_bool(&param_server, "publish_toogle", true);
    //rclc_parameter_set_int(&param_server, "publish_rate_ms", 1000);

	Log.println("before spin");
  	RCCHECK(rclc_executor_spin(&executor));
	Log.println("after spin");

	RCCHECK(rcl_publisher_fini(&publisher, &node));
	RCCHECK(rcl_node_fini(&node));
}

```

</p>
</details>

<details><summary>platformio.ini</summary>
<p>

```ini
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
upload_protocol = teensy-cli
board_microros_transport = serial
board_microros_distro = galactic
board_microros_user_meta = project_colcon.meta
lib_deps = https://github.com/micro-ROS/micro_ros_platformio

```

</p>
</details>

<details><summary>project_colcon.meta</summary>
<p>

```python
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=20",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=20",
                "-DRMW_UXRCE_MAX_SERVICES=10",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=2",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```

</p>
</details>

3. Compile & flash teensy
4. Setup and Start microros agent 

#### Expected behavior

Executer spins, counter is published and parameters are available by calling `ros2 param list`.

#### Actual behavior

Output: `Failed status on line 84: 1. Aborting.` which is:

```c
// Create parameter service
RCCHECK(rclc_parameter_server_init_default(&param_server, &node));
```

Acuadros95, 2022-06-13:
Hi @fischejo, 
You need to update the memory configuration on the `colcon.meta` file, or create your own one as explained `here`.

For a parameter server, you will need at least 1 publisher and 5 services, check this tutorial section: [Memory requirements](https://micro.ros.org/docs/tutorials/programming_rcl_rclc/parameters/#memory-requirements)

Acuadros95, 2022-06-13:
What is the Agent output on debug mode `-v6`?
Anyway, I will try to replicate this on our side.

Acuadros95, 2022-06-13:
Here is the fix: https://github.com/micro-ROS/micro_ros_platformio/pull/34

fischejo, 2022-06-13:
Awesome. Thank you for the fast fixing. It works now :+1: .

closed 2022-06-13 by fischejo