micro-ROS/micro_ros_platformio #38: RP2040

samuk, 2022-07-15:
## Issue template

- Hardware description: RP2040
- RTOS: 
- Installation type: 
- Version or commit hash: humble

#### Steps to reproduce the issue

Using https://github.com/Agroecology-Lab/linorobot2_hardware/blob/galactic/firmware/platformio.ini
Follow https://github.com/linorobot/linorobot2_hardware#3-install-platformio

#### Expected behavior
pio builds firmware

#### Actual behavior

--- stderr: ament_cmake_core
Traceback (most recent call last):
  File "/home/ubuntu/linorobot2_hardware/calibration/.pio/libdeps/pico/micro_ros_platformio/build/dev/src/ament_cmake/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at ament_cmake_package_templates-extras.cmake:41 (message):

#### Additional information
https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk
https://gist.github.com/Redstone-RM/0ca459c32ec5ead8700284ff56a136f7

pablogs9, 2022-07-18:
Could you please provide details for replicating your exact environment? Could you share your specific code, `.ini``file, replicable project?

samuk, 2022-07-18:
Sure, this is the .ini file I'm using: https://github.com/Agroecology-Lab/linorobot2_hardware/blob/galactic/firmware/platformio.ini

samuk, 2022-07-18:
sudo apt-get -y install ament-cmake

Has resolved this issue

closed 2022-07-18 by samuk