micro-ROS/micro_ros_platformio #42: Adding custom ROS messages for microROS platformIO

Genozen, 2022-07-21:
I'm trying to follow the tutorial from microROS custom messages: [](https://micro.ros.org/docs/tutorials/advanced/create_new_type/)

Is there a similar way to do this custom messages in the platformIO version?

pablogs9, 2022-07-22:
Just copy your packages to the folder mentioned here: https://github.com/micro-ROS/micro_ros_platformio#extra-packages

closed 2022-07-22 by pablogs9

ssaattww, 2022-07-22:
is it means  "myProject/.pio/libdeps/board_name/micro_ros_platformio/extra_packages/my_packages"?

pablogs9, 2022-07-22:
That's it

ssaattww, 2022-07-22:
No need to add parent directry to libdeps like this, right? https://github.com/micro-ROS/micro_ros_platformio/blob/c9c410c9c7b93cca685cd19786978b0589f23ad9/ci/platformio.ini#L9

pablogs9, 2022-07-22:
I don't think so. Do we have any examples of custom packages for platformio @Acuadros95 ?

Acuadros95, 2022-07-22:
You can check out our CI, which includes a project example with `extra_packages`: [link](https://github.com/micro-ROS/micro_ros_platformio/tree/main/ci).

> is it means "myProject/.pio/libdeps/board_name/micro_ros_platformio/extra_packages/my_packages"?

The folder should be at the same level of the `platformio.ini` file: `myProject/extra_packages/my_packages`, sorry for the confusion.

ssaattww, 2022-07-23:
solved. Thank you for your support.

Genozen, 2022-07-25:
@ssaattww  did you get this error?
```
  // // create publisher2
  RCCHECK(rclc_publisher_init_default(
    &publisher2,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(my_custom_message, custom_msg, MyCustomMessage),
    "micro_ros_my_custom_publisher"));
```
[image]

ssaattww, 2022-08-03:
no, Once I success to built, all is done.