micro-ROS/micro_ros_platformio #43: Handling String type Publisher messege

Genozen, 2022-07-22:
## Issue template
I'm trying to publish String type message, but having some trouble doing it in the microROS version.

- Hardware description: ESP32 Heltec LoRa v2
- Installation type: microROS PlatformIO
- Version or commit hash: Linux 20.04 with Foxy ROS2

#### Steps to reproduce the issue
`platformIO.ini`
```
[env:esp32doit-devkit-v1]
; [env: heltec_wifi_lora_32_V2]
platform = espressif32
board = esp32doit-devkit-v1
; board = heltec_wifi_lora_32_V2
board_microros_transport = wifi ;serial ;wifi ;wifi_nina
board_microros_distro = foxy
framework = arduino
monitor_speed = 115200
lib_deps =
    adafruit/Adafruit BusIO @ ^1.12.0
    sparkfun/SparkFun u-blox GNSS Arduino Library @ ^2.2.10
    https://github.com/micro-ROS/micro_ros_platformio # for microROS
    stevemarple/SoftWire @ ^2.0.9
    stevemarple/AsyncDelay @ ^1.1.2
    adafruit/RTClib @ ^2.0.3
```

`main.cpp`
```
#include <Arduino.h>
#include <micro_ros_platformio.h>

#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
// #include <micro_ros_utilities/type_utilities.h>
// #include <micro_ros_utilities/string_utilities.h>

#include <std_msgs/msg/int32.h> //for integer messages
#include <std_msgs/msg/string.h>

#include "freertos/FreeRTOS.h"      // freeRTOS Libraries 
#include "freertos/task.h"          // freeRTOS Task Header
#include "esp_system.h"
#include "soc/rtc_wdt.h"   
#include <mySD.h>
#include <SparkFun_u-blox_GNSS_Arduino_Library.h>

//publisher initialization
rcl_publisher_t publisher;
rcl_publisher_t publisher2;
std_msgs__msg__Int32 msg;
std_msgs__msg__String msg_gps;

//rclc initialization...
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

// Error handle loop
void error_loop() {
  while(1) {
    delay(100);
  }
}

//This is the function where the message data is set and the msg are actually published. RCLCPP_INFO macro ensures every published msg is printed to the console
void timer_callback(rcl_timer_t * timer, int64_t last_call_time) {
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));

    msg.data+=1;
  }
}

void setup() {
  // ----------- Configure serial transport
  Serial.begin(115200);
  // set_microros_serial_transports(Serial);

  //***** For Router WiFi....
  IPAddress agent_ip(---, ---, --, --);
  size_t agent_port = 8888;
  char ssid[] = "----------";
  char psk[]= "----------";

  set_microros_wifi_transports(ssid, psk, agent_ip, agent_port);

  //=========================ROS2 stuff setup..............===================
  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_platformio_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "micro_ros_platformio_node_publisher"));

  // create publisher2
  RCCHECK(rclc_publisher_init_default(
    &publisher2,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_gps, String),
    "micro_ros_platformio_node_publisher_lat"));

  // micro_ros_utilities_create_message_memory(
  //   ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_gps, String),
  //   &msg_gps,
  //   (micro_ros_utilities_memory_conf_t) {}
  // );

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 0;
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

```

`Error message: pio run`
```
src/main.cpp: In function 'void setup()':
.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/libmicroros/include/rosidl_runtime_c/message_type_support_struct.h:78:5: error: 'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg_gps__String' was not declared in this scope
     rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
     ^~~~~~~~~~~~~~~~~~~~
src/main.cpp:39:43: note: in definition of macro 'RCCHECK'
 #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
                                           ^~
.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/libmicroros/include/rosidl_typesupport_interface/macros.h:27:3: note: in expansion of macro 'ROSIDL_TYPESUPPORT_INTERFACE__SYMBOL_NAME'
   ROSIDL_TYPESUPPORT_INTERFACE__SYMBOL_NAME( \
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/libmicroros/include/rosidl_runtime_c/message_type_support_struct.h:77:3: note: in expansion of macro 'ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME'
   ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME( \
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:277:5: note: in expansion of macro 'ROSIDL_GET_MSG_TYPE_SUPPORT'
     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_gps, String),
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/libmicroros/include/rosidl_runtime_c/message_type_support_struct.h:78:5: note: suggested alternative: 'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__String'
     rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
     ^~~~~~~~~~~~~~~~~~~~
src/main.cpp:39:43: note: in definition of macro 'RCCHECK'
 #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
                                           ^~
.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/libmicroros/include/rosidl_typesupport_interface/macros.h:27:3: note: in expansion of macro 'ROSIDL_TYPESUPPORT_INTERFACE__SYMBOL_NAME'
   ROSIDL_TYPESUPPORT_INTERFACE__SYMBOL_NAME( \
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/libmicroros/include/rosidl_runtime_c/message_type_support_struct.h:77:3: note: in expansion of macro 'ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME'
   ROSIDL_TYPESUPPORT_INTERFACE__MESSAGE_SYMBOL_NAME( \
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:277:5: note: in expansion of macro 'ROSIDL_GET_MSG_TYPE_SUPPORT'
     ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_gps, String),
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/esp32doit-devkit-v1/src/main.cpp.o] Error 1
```

#### Expected behavior
Publishing String type messege

#### Actual behavior

#### Additional information

Genozen, 2022-07-22:
Added some more stuff:

In setup:
```
  micro_ros_utilities_create_message_memory(
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String),
    &msg_array_string,
    (micro_ros_utilities_memory_conf_t) {}
  );
```

In main loop:
```
  msg_array_string.data = micro_ros_string_utilities_set(msg_array_string.data, "Stationary");
```

Genozen, 2022-07-23:
[image]

pablogs9, 2022-07-26:
Could you try setting the transport to only one? Instead of `board_microros_transport = wifi ;serial ;wifi ;wifi_nina`, something like `board_microros_transport = wifi`.

Also, could you share the whole build log?

Acuadros95, 2022-07-26:
You have a typo in `create publisher2`: `ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_gps, String),`
The second argument should be `msg`. 

Please take care when using find and replace.

pablogs9, 2022-08-04:
Any update on this?

Genozen, 2022-08-04:
I haven't had the chance to try this deeper. I've fixed the typos and will give an update next week.

Genozen, 2022-08-11:
Closing this for now, since I won't be using it. But whoever is trying this just ensure you initialized the fields in the data struct given. and should be good.

closed 2022-08-11 by pablogs9