micro-ROS/micro_ros_platformio #46: Allow freezing the micro-ROS package versions

bjsowa, 2022-08-01:
I got back to working on my project after a couple of months. When I tried to build it, it no longer compiled because the library builder pulled newer version of micro-ROS packages that contained API changes that were not compatible with my code.

I think this library should either be tied to specific versions of the packages and have a library auto-update workflow (similar to the on [micro_ros_arduino](https://github.com/micro-ROS/micro_ros_arduino)) or allow the user to specify the versions.

pablogs9, 2022-08-01:
[Here](https://github.com/micro-ROS/micro_ros_platformio/blob/b256be95073d01fd06ba5f97dbb7c312e3dc697c/microros_utils/repositories.py#L105) is a good point to specify fixed commits or tags for each micro-ROS package. Please open a PR if you have any solution.

patrickwasp, 2023-02-23:
you can try using the version in the URL in your lib_deps

```
lib_deps = 
	https://github.com/micro-ROS/micro_ros_platformio#1.0.0
```