micro-ROS/micro_ros_platformio #50: Compilation problem when adding tf2 package

gggard, 2022-08-06:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: 
- Version or commit hash: galactic

#### Steps to reproduce the issue
I would like to add tf2 package to the build.
I have updated extra_packages.repo as such :
```
repositories:
  tf2:
    type: git
    url: https://github.com/ros2/geometry2
    version: galactic
```

I cleaned the build and made a pio run

#### Expected behavior

Build OK

#### Actual behavior

The compilation failed complaining about hpp (geom_msg) files not found in the code of tf2. Indeed they are present with the .h extension but not the hpp extension.

Trying to patch tf2 to replace hpp by h include failed also (complain about the namespace usage of tf2).

It seams that idl is producing c file and not cpp files, any clue ? or something I miss ?
Thanks in advance !

pablogs9, 2022-08-08:
https://github.com/ros2/geometry2 has a lot of packages, which one do you want to use?

pablogs9, 2022-09-19:
Any update on this?

Krizzel87, 2022-09-25:
Error while compiling with extra_packages.repos:
```
repositories:
  control_msgs:
    type: git
    url: https://github.com/ros-controls/control_msgs
    version: humble

  geometry2:
    type: git
    url: https://github.com/ros2/geometry2
    version: humble

```
Get Error:
```

--- stderr: tf2
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findconsole_bridge_vendor.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "console_bridge_vendor", but CMake did not find one.

  Could not find a package configuration file provided by
  "console_bridge_vendor" with any of the following names:

    console_bridge_vendorConfig.cmake
    console_bridge_vendor-config.cmake

  Add the installation prefix of "console_bridge_vendor" to CMAKE_PREFIX_PATH
  or set "console_bridge_vendor_DIR" to a directory containing one of the
  above files.  If "console_bridge_vendor" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< tf2 [0.57s, exited with code 1]
```
Tried Ros2 Humble, but still not work with geometry2.
The controll message works fine.

Krizzel87, 2022-09-29:
i want to use tf2

pablogs9, 2022-09-30:
tf2 depends on cpp layers and it is not directly portable to micro-ROS. You will need to port it or write a compatible implementation

closed 2022-09-30 by pablogs9