micro-ROS/micro_ros_platformio #51: Ethernet support for ESP32

fabianhick, 2022-08-11:
- Hardware description: Olimex ESP32-GATEWAY

I'm currently trying to implement support of Ethernet for the ESP32 with microROS, however, the ESP-IDF API looks really complicated, so I tried to use https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet in conjunction with this project.

However, it isn't working as expected:
```
E (1014) esp.emac: emac_esp32_init(349): reset timeout
E (1015) esp_eth: esp_eth_driver_install(214): init mac failed
[  1018][E][ETH.cpp:323] begin(): esp_eth_driver_install failed
[  4079][E][WiFiUdp.cpp:183] endPacket(): could not send data: 118
```

Do you have any tips or ideas on how to implement Ethernet support for the ESP32 into microROS?

pablogs9, 2022-08-11:
Not related to micro-ROS

closed 2022-08-11 by pablogs9

fabianhick, 2022-08-11:
Is it not planned to extend micro-ROS to allow Ethernet as a transport layer with an ESP32 just like with the Teensy 4.1?

pablogs9, 2022-08-11:
If you have a valid transport layer for micro-ROS using Ethernet and Platform.IO please contribute with a PR