micro-ROS/micro_ros_platformio #55: Could not find ROS middleware implementation

scch1092, 2022-08-29:
## Issue template

- Hardware description: Ubuntu 20.04, teensy 4.1
- RTOS: Arduino
- Installation type: VS-Code + PlatformIO (extension for VS-Code)
- Version or commit hash: galactic

#### Steps to reproduce the issue
Step 1: Copy the contents of the micro_ros_publisher example.
Step 2: platform.ini
```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino

board_microros_distro = galactic
board_microros_transport = serial
board_microros_user_meta = colcon.meta

lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio.git

```
Step3: pio lib install
Step4: pio run

#### Expected behavior
Build the packages

#### Actual behavior
Build fails with the following output:
```
--- stderr: rmw_implementation
CMake Error at /home/scch1092/example/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake:60 (message):
  Could not find ROS middleware implementation 'rmw_fastrtps_cpp'.  Choose
  one of the following: rmw_microxrcedds
Call Stack (most recent call first):
  CMakeLists.txt:22 (get_default_rmw_implementation)

---
Failed   <<< rmw_implementation [0.44s, exited with code 1]
```

#### Additional information
```
$ env | grep ROS

ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=galactic
```
```
$ env | grep RMW

RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```
```
env | grep AMENT_PREFIX_PATH

AMENT_PREFIX_PATH=/home/scch1092/ros2_ws/install/micro_ros_agent:/home/scch1092/ros2_ws/install/micro_ros_msgs:/opt/ros/galactic
```

Acuadros95, 2022-08-29:
> RMW_IMPLEMENTATION=rmw_fastrtps_cpp

Do you have fast rtps installed? What is your output for `apt list --installed | grep fastrtps`?

Anyway, you can manually set `export RMW_IMPLEMENTATION=rmw_microxrcedds` to build micro-ROS.

scch1092, 2022-08-29:
```
apt list --installed | grep fastrtps

ros-galactic-fastrtps-cmake-module/focal,now 1.2.1-2focal.20220430.052920 amd64 [installed,automatic]
ros-galactic-fastrtps/focal,now 2.3.4-1focal.20220430.111237 amd64 [installed,automatic]
ros-galactic-rmw-fastrtps-cpp/focal,now 5.0.2-1focal.20220729.231320 amd64 [installed,automatic]
ros-galactic-rmw-fastrtps-shared-cpp/focal,now 5.0.2-1focal.20220729.230057 amd64 [installed,automatic]
ros-galactic-rosidl-typesupport-fastrtps-c/focal,now 1.2.1-2focal.20220726.200223 amd64 [installed,automatic]
ros-galactic-rosidl-typesupport-fastrtps-cpp/focal,now 1.2.1-2focal.20220726.195810 amd64 [installed,automatic]

```

> Anyway, you can manually set export RMW_IMPLEMENTATION=rmw_microxrcedds to build micro-ROS.

I will try it, thanks!

scch1092, 2022-08-29:
> Anyway, you can manually set export RMW_IMPLEMENTATION=rmw_microxrcedds to build micro-ROS.

No luck, still the same problem

pablogs9, 2022-09-05:
Could you try to build a micro-ROS environment without sourcing the ROS 2 installation?

Acuadros95, 2022-09-09:
Hi @scch1092,

This PR should fix your issue: https://github.com/micro-ROS/micro_ros_platformio/pull/62
Please check it out and provide feedback

scch1092, 2022-09-09:
Hello, thank you for the feedback. It seems that it is probably my computer. On another computer I was able to create the libraries and flash them to the MCU.

Acuadros95, 2022-09-09:
The problem was related to setting `RMW_IMPLEMENTATION=rmw_fastrtps_cpp` by default.

The PR should fix the build process in that case, you should be able to compile in your computer.

Closing!

closed 2022-09-09 by Acuadros95