micro-ROS/micro_ros_platformio #57: Recent upload gives "alternative suggestion: try using serial"

Genozen, 2022-09-01:
- Hardware description: ESP32 Heltech LoRa V2
- RTOS: Linux
- Installation type: microROS platformIO with Arduino framework
- Version or commit hash: Foxy

#### Steps to reproduce the issue
```void setup() {
  // ----------- Configure serial transport
  Serial.begin(115200);
  set_microros_serial_transports(Serial);
}```

#### Expected behavior
Runs in serial

#### Actual behavior
A yellow message suggesting me to run it as WiFi

#### Additional information
I find it weird as the compilation to the board seems to be "lagging behind"?

When I set to utilize serial, the compilation gives a yellow suggestion saying something like alternative suggestion: try using WiFi.
And when I switched to utilize WiFi, the yellow message gives the alternative suggestion saying to try using Serial...?

Code works about a month ago and I never touched it until now.

Acuadros95, 2022-09-02:
Could not replicate this, could you share your `platformio.ini` and your source code?

Does this affects functionality?

pablogs9, 2022-09-19:
Any update on this?

Acuadros95, 2022-10-20:
Closing due to inactivity

closed 2022-10-20 by Acuadros95