micro-ROS/micro_ros_platformio #59: "VisualStudioVersion is not set, please run within a Visual Studio Command Prompt."

SpMakris, 2022-09-07:
I am trying to install the micro_ros_platformio library but colcon build fails.
Platformio version: 6.1.4
Visual Studio Code version: 1.7.1.0
OS: Windows 10

platfomrio.ini :
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:dueUSB]
platform = atmelsam
board = dueUSB
framework = arduino

lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
	jandrassy/EthernetENC@^2.0.2

monitor_speed = 115200
```

Building the project results in the following output:

```

Processing dueUSB (platform: atmelsam; board: dueUSB; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/dueUSB.html
PLATFORM: Atmel SAM (8.1.0) > Arduino Due (USB Native Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduino-sam @ 1.6.12
 - framework-cmsis @ 1.40500.0 (4.5.0)
 - framework-cmsis-atmel @ 1.2.2
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
C:\Users\spiro\.platformio\penv\Scripts\python.exe -m pip install pyyaml
Requirement already satisfied: pyyaml in c:\users\spiro\.platformio\penv\lib\site-packages (6.0)

[notice] A new release of pip available: 22.1.2 -> 22.2.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Installing markupsafe==2.0.1 with pip at PlatformIO environment
C:\Users\spiro\.platformio\penv\Scripts\python.exe -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in c:\users\spiro\.platformio\penv\lib\site-packages (2.0.1)

[notice] A new release of pip available: 22.1.2 -> 22.2.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Configuring dueUSB with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Build dev micro-ROS environment failed: 
 --- stderr: gtest_vendor

Traceback (most recent call last):

  File "C:\Users\spiro\.platformio\penv\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__

    rc = await self.task(*args, **kwargs)

  File "C:\Users\spiro\.platformio\penv\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__

    return await task_method(*args, **kwargs)

  File "C:\Users\spiro\.platformio\penv\lib\site-packages\colcon_ros\task\cmake\build.py", line 35, in build

    rc = await extension.build(

  File "C:\Users\spiro\.platformio\penv\lib\site-packages\colcon_cmake\task\cmake\build.py", line 87, in build

    rc = await self._reconfigure(args, env)

  File "C:\Users\spiro\.platformio\penv\lib\site-packages\colcon_cmake\task\cmake\build.py", line 153, in _reconfigure

    raise RuntimeError(

RuntimeError: VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.

---

Failed   <<< gtest_vendor [0.89s, exited with code 1]

VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.
```

I have tried the exact same process in Ubuntu 20 and didn't get any errors.

pablogs9, 2022-09-07:
This module is not available for Windows platforms.

closed 2022-09-07 by pablogs9

SpMakris, 2022-09-07:
> This module is not available for Windows platforms.

Shouldn't this be reflected in the documentation? Previous versions (before micro_ros_arduino was split) used to work fine on Windows, so that's not an unreasonable assumption.

pablogs9, 2022-09-07:
Sorry, yes, this should be reflected in the documentation, we will update it. 

micro-ROS for Arduino work in Linux since it provides a set of precompiled libraries and generated headers ready to be linked with Arduino applications. This module actually builds the micro-ROS library on-the-fly along with the platform.io code. We have had no time to investigate how to make our build procedure compatible with Linux and Windows simultaneously so, in general, all micro-ROS components and modules are Linux-ony.

Sorry for the inconvenience and if you have any proposal on how to make this happen in Windows, we are completely open to PR.