micro-ROS/micro_ros_platformio #60: ament_cmake clone failed:  fatal: Too many arguments.

matthew-william-lock, 2022-09-08:
## Issue template

Having trouble building the micro-ros

```
Downloading micro-ROS dev dependencies
ament_cmake clone failed: 
fatal: Too many arguments.

usage: git clone [<options>] [--] <repo> [<dir>]

    -v, --verbose         be more verbose
    -q, --quiet           be more quiet
    --progress            force progress reporting
    --reject-shallow      don't clone shallow repository
    -n, --no-checkout     don't create a checkout
    --bare                create a bare repository
    --mirror              create a mirror repository (implies bare)
    -l, --local           to clone from a local repository
    --no-hardlinks        don't use local hardlinks, always copy
    -s, --shared          setup as shared repository
    --recurse-submodules[=<pathspec>]
                          initialize submodules in the clone
    --recursive ...       alias of --recurse-submodules
    -j, --jobs <n>        number of submodules cloned in parallel
    --template <template-directory>
                          directory from which templates will be used
    --reference <repo>    reference repository
    --reference-if-able <repo>
                          reference repository
    --dissociate          use --reference only while cloning
    -o, --origin <name>   use <name> instead of 'origin' to track upstream
    -b, --branch <branch>
                          checkout <branch> instead of the remote's HEAD
    -u, --upload-pack <path>
                          path to git-upload-pack on the remote
    --depth <depth>       create a shallow clone of that depth
    --shallow-since <time>
                          create a shallow clone since a specific time
    --shallow-exclude <revision>
                          deepen history of shallow clone, excluding rev
    --single-branch       clone only one branch, HEAD or --branch
    --no-tags             don't clone any tags, and make later fetches not to follow them
    --shallow-submodules  any cloned submodules will be shallow
    --separate-git-dir <gitdir>
                          separate git dir from working tree
    -c, --config <key=value>
                          set config inside the new repository
    --server-option <server-specific>
                          option to transmit
    -4, --ipv4            use IPv4 addresses only
    -6, --ipv6            use IPv6 addresses only
    --filter <args>       object filtering
    --also-filter-submodules
                          apply partial clone filters to submodules
    --remote-submodules   any cloned submodules will use their remote-tracking branch
    --sparse              initialize sparse-checkout file to include only files at root

========================================================================================== [FAILED] Took 4.28 seconds ==========================================================================================

```

- Hardware description: Teensy 4.1, developing on windows 11

config file

```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_distro = humble
board_microros_transport = serial
lib_deps = 
    https://github.com/micro-ROS/micro_ros_platformio
```

#### Steps to reproduce the issue

Click on the build button in VS code

Acuadros95, 2022-09-09:
This module is not currently available for Windows platforms. 

This may change in the future, but its not in our development plans at this moment.

closed 2022-09-09 by Acuadros95

matthew-william-lock, 2022-09-09:
Hey @Acuadros95 

Sorry I didn't fully specify my environment.

I am on WSL2 using Ubuntu 22.04

Acuadros95, 2022-09-09:
What is your git version?

simonllopez, 2023-07-24:
have the same error with Ubuntu 22.04.2, git version 2.34.1.

I need to copy the 'libdeps' folder from another Ubuntu distribution in order to build the project.