micro-ROS/micro_ros_platformio #61: Could not find a package configuration file provided by   "rosidl_runtime_cpp" with any of the following names:

matthew-william-lock, 2022-09-09:
- Hardware description: trying to target  teensy 4.1 and building on Ubuntu 22.04 (native install)

Config file : 
```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_transport = serial
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```

# Error

```
Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/matt/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/matt/.platformio/penv/lib/python3.10/site-packages (6.0)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/matt/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/matt/.platformio/penv/lib/python3.10/site-packages (2.0.1)
Configuring teensy41 with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Downloading micro-ROS library
         - Downloaded microcdr
         - Downloaded microxrcedds_client
         - Downloaded rcl_action
         - Downloaded rcl_yaml_param_parser (ignored)
         - Downloaded rcl_lifecycle
         - Downloaded rcl
         - Downloaded rclc_parameter
         - Downloaded rclc
         - Downloaded rclc_lifecycle
         - Downloaded rclc_examples (ignored)
         - Downloaded micro_ros_utilities
         - Downloaded rcutils
         - Downloaded micro_ros_msgs
         - Downloaded rmw_microxrcedds
         - Downloaded rosidl_typesupport_c
         - Downloaded rosidl_typesupport_cpp (ignored)
         - Downloaded rosidl_typesupport_microxrcedds_cpp (ignored)
         - Downloaded rosidl_typesupport_microxrcedds_c
         - Downloaded rosidl_typesupport_microxrcedds_c_tests
         - Downloaded rosidl_typesupport_microxrcedds_cpp_tests
         - Downloaded rosidl_typesupport_microxrcedds_test_msg
         - Downloaded rosidl_typesupport_introspection_cpp (ignored)
         - Downloaded rosidl_typesupport_interface
         - Downloaded rosidl_adapter
         - Downloaded rosidl_cmake
[... 219 lines cut ...]

    rosidl_runtime_cppConfig.cmake
    rosidl_runtime_cpp-config.cmake

  Add the installation prefix of "rosidl_runtime_cpp" to CMAKE_PREFIX_PATH or
  set "rosidl_runtime_cpp_DIR" to a directory containing one of the above
  files.  If "rosidl_runtime_cpp" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/matt/microros_ws/install/rosidl_typesupport_microxrcedds_cpp/share/rosidl_typesupport_microxrcedds_cpp/cmake/rosidl_typesupport_microxrcedds_cppConfig.cmake:41 (include)
  /home/matt/projects/Embedded-UAV-Cooperative-Localisation/src/teensy/simple-publisher/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/matt/projects/Embedded-UAV-Cooperative-Localisation/src/teensy/simple-publisher/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< unique_identifier_msgs [0.57s, exited with code 1]

======================================================================= [FAILED] Took 67.58 seconds =======================================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

#### Steps to reproduce the issue
Build the sample publisher package

matthew-william-lock, 2022-09-09:
Okay I managed to get it working...

I had to remove my own ```microros_ws``` folder in my root directory.

Is there no way to have my own micro_ros directory alongside the one platform io is trying to use?

Acuadros95, 2022-09-09:
> Is there no way to have my own micro_ros directory alongside the one platform io is trying to use?

What does your directory look like? PlatformIO is not downloading and building micro-ROS inside the `.pio` folder?

Acuadros95, 2022-09-09:
Closing since its solved

closed 2022-09-09 by Acuadros95