micro-ROS/micro_ros_platformio #64: compiling issue

usaotearoa, 2022-09-24:
Issue compiling in Visual Studio Code with PlatformIO.
Created blank project, added the config in platfomio.ini and copied the example.
Getting note that there is an update for pip

`notice] A new release of pip available: 22.1.2 -> 22.2.2
[notice] To update, run: python.exe -m pip install --upgrade pip`

Therefore tried to update pip, but it says its already latest:
`D:\documents\programming\git_ws\AutoBot_uros>python.exe -m pip install --upgrade pip
Requirement already satisfied: pip in c:\users\admin\appdata\local\programs\python\python310\lib\site-packages (22.2.2)`
?? 

Of course also tried to do a clean, and tried again. Python is installed, path variables are set-up. Not sure why it is failing.

Also tried to manually set the VS version to 17.0, as that seems to be one of the supported versions, but then it fails some place else, as the arguments for CMake then don't make sense anylonger.

Not sure what I am missing here, been trying to find a solution on google and stackoverflow, also checked the issues list here, but couldn't find anything. 

If anyone is out there reading this, please help me getting this to compile. I'm trying to use this weekend to make progress on my project, but so far only issues and no real progress made :-/

Here the entire output trace:
`Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.16.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash 
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy @ 1.156.0 (1.56)
 - tool-teensy @ 1.155.0 (1.55)
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
C:\Users\Admin\.platformio\penv\Scripts\python.exe -m pip install pyyaml
Requirement already satisfied: pyyaml in c:\users\admin\.platformio\penv\lib\site-packages (6.0)

[notice] A new release of pip available: 22.1.2 -> 22.2.2       
[notice] To update, run: python.exe -m pip install --upgrade pip
Installing markupsafe==2.0.1 with pip at PlatformIO environment
C:\Users\Admin\.platformio\penv\Scripts\python.exe -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in c:\users\admin\.platformio\penv\lib\site-packages (2.0.1)

[notice] A new release of pip available: 22.1.2 -> 22.2.2       
[notice] To update, run: python.exe -m pip install --upgrade pip
Configuring teensy41 with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Build dev micro-ROS environment failed: 
 --- stderr: gtest_vendor

Traceback (most recent call last):

  File "C:\Users\Admin\.platformio\penv\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__

    rc = await self.task(*args, **kwargs)

  File "C:\Users\Admin\.platformio\penv\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__

    return await task_method(*args, **kwargs)

  File "C:\Users\Admin\.platformio\penv\lib\site-packages\colcon_ros\task\cmake\build.py", line 35, in build

    rc = await extension.build(

  File "C:\Users\Admin\.platformio\penv\lib\site-packages\colcon_cmake\task\cmake\build.py", line 87, in build

    rc = await self._reconfigure(args, env)

  File "C:\Users\Admin\.platformio\penv\lib\site-packages\colcon_cmake\task\cmake\build.py", line 153, in _reconfigure

    raise RuntimeError(

RuntimeError: VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.

---

Failed   <<< gtest_vendor [1.25s, exited with code 1]

VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.

================================================================================================================================================================================================================================== [FAILED] Took 28.90 seconds ==================================================================================================================================================================================================================================

 *  The terminal process "C:\Users\Admin\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

pablogs9, 2022-09-26:
@Acuadros95 can you take a look?

Acuadros95, 2022-09-26:
What is your version of platformIO?
Also, share your `platform.ini` so I can try to replicate.

usaotearoa, 2022-10-01:
Needed to follow the proper PlatformIO build workflow:
<CLEAN> always a good idea. Sometimes I manually delete the ".pio" directory and then follow as below
pio lib install # Install dependencies
pio run # Build the firmware
pio run --target upload # Flash the firmware

closed 2022-10-01 by usaotearoa