micro-ROS/micro_ros_platformio #66: porting issue addtwoints service

usaotearoa, 2022-10-01:
Using teensy 4.1 with platformio in VSCode on Ubuntu 22.04

I have followed the tutorial:
https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/

Now I'm trying to expand this for the  "micro-ros_addtwoints_servcie".
However I'm having issue understanding how to port this over. I do understand this example comes from the pure "arduino" microROS side and there are a few differences. But what I lack is the understanding how this could be ported over.

Currently it already hangs at the point where I 
#include <example_interfaces/srv/add_two_ints.h>

Please help me understand how I can port this over.
Once I have a working example I will update the github repo, and add this add_two_ints so others can also benefit.

Thanks

usaotearoa, 2022-10-01:
Do I maybe have to use the 
https://github.com/micro-ROS/micro-ROS-demos/tree/foxy/rclc/addtwoints_server)/main.c
and port that over to platformio?

How would I do that?

usaotearoa, 2022-10-01:
Another question is,
Do I need to do anything in terms of FreeRTOS, when going the approach of using micro-ROS with the teensy under platformio on Ubuntu? Or is this all handled in the background?

pablogs9, 2022-10-03:
> Currently it already hangs at the point where I
> #include <example_interfaces/srv/add_two_ints.h>

Which is the error? As far as I see, you have this file [here](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/src/example_interfaces/srv/add_two_ints.h).

> Do I need to do anything in terms of FreeRTOS, when going the approach of using micro-ROS with the teensy under platformio on Ubuntu? Or is this all handled in the background?

Arduino + Teensy port of micro-ROS does not use FreeRTOS. It's a bare metal approach.

usaotearoa, 2022-10-07:
I'm trying to port parts from 
`micro_ros_arduino/tree/humble/examples)/micro-ros_addtwoints_service/` to work with the Teensy 4.1 in PlatformIO

However I'm facing issues with it along the way.

The problem is with the following line of code 
`// create service
  // Get message type support
  const rosidl_message_type_support_t * type_support = ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts);`
It doesn't know `ROSIDL_GET_SRV_TYPE_SUPPORT`

Here is the entire source code main.cpp:
`#include <Arduino.h>
#include <micro_ros_platformio.h>
#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/int64.h>
//#include <example_interfaces/srv/add_two_ints.h>

#if !defined(MICRO_ROS_TRANSPORT_ARDUINO_SERIAL)
#error This example is only avaliable for Arduino framework with serial transport.
#endif

// Publisher object
rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;

// Service server object
rcl_service_t service;
const char * service_name = "/addtwoints";

// define rclc executor
rclc_executor_t executor;

// support object
rclc_support_t support;

// micro-ROS allocator object
rcl_allocator_t allocator;

// create node object
rcl_node_t node;
// Create and initialize timer object
rcl_timer_t timer;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

// Error handle loop
void error_loop() {
  while(1) {
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time) {
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

void setup() {
  // Configure serial transport
  Serial.begin(115200);
  set_microros_serial_transports(Serial);
  delay(2000);

  // Initialize micro-ROS allocator
  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_platformio_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(&publisher,&node,ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),"micro_ros_platformio_node_publisher"));

  // create service
  // Get message type support
  const rosidl_message_type_support_t * type_support = ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts);

  // create timer,
  const unsigned int timer_timeout = 1000;

  // Initialize support object
  RCCHECK(rclc_timer_init_default(&timer,&support,RCL_MS_TO_NS(timer_timeout),timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 0;
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}`

And also here is my "platformio.ini"
`[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_transport = serial

lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio`

Any suggestion, why that line is not recognized? What am I missing here?

pablogs9, 2022-10-10:
Hello, I'm closing this since you have opened multiple issues with the same problem: https://github.com/micro-ROS/micro_ros_platformio/issues/68

Please try to be patient and open only one issue. If you need to ping us, please use the Slack channel.
As we mention in the other issue, example_interfaces package shall be added to PlatformIO

closed 2022-10-10 by pablogs9