micro-ROS/micro_ros_platformio #67: Micro-ros undefined references

thechosenone98, 2022-10-03:
I am trying to build micro_ros_platformio for a Teensy4.1 but I get a lot of warnings from "ar" at the end of the build but no fatal error. Here is a picture of those warning.

[image]

Further in the build of the code itself (the code that depends on micro_ros), it fails due to a bunch of undefined references that correspond with the file that are mentionned by the "ar" command.

#### Steps to reproduce the issue
- OS: MacOS 12.5
- Clone this repo: [Teensy Vs. Upduino Particle Filter](https://github.com/matthew-william-lock/teensy-vs-upduino-particle-filter)
- Open VS Code in the folder src/teensy_particle_filter
- Build through PlatformIO
- That's it

#### Expected behavior

Build without errors (it compiles without all the aforementioned warnings on another machine running Ubuntu)

#### Actual behavior

Throws some warnings and complains later in the build that there are undefined references

#### Additional information
If it compiles without the warning, there will be an error at the end relating to the random library but this is due to other unrelated stuff.

pablogs9, 2022-10-04:
The build system for micro-ROS in Platform.io does not work properly in macOS or Windows, it is designed for Linux.

AFAIK you are able to build it with an Ubuntu machine. Sorry for the inconvenience, if you manage to solve these issues and support macOS we are open to PRs.

Thanks!

usaotearoa, 2022-10-08:
I'm facing a very similar issue, despite being on Ubuntu 22.04.1. Here are the details:
Ubuntu 22.04.1 with ROS2 Humble install (desktop)
Visual Studio Code Version: 1.71.2
PlatformIO Core 6.1.4·Home 3.4.3 
Hardware Platform Teensy 4.1 and micro_ROS (platformio)

While attempting to port source code from the following to work on teensy in platformio I have encounter the below compiler issues.
Porting from:
Mirco_ROS_Arduino - micro-ros_addtwoints_service
to Teensy 4.1 with serial communication:

The source code:
```
//#include <micro_ros_arduino.h>
#include <Arduino.h>
#include <micro_ros_platformio.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t service;
rcl_wait_set_t wait_set;

example_interfaces__srv__AddTwoInts_Response res;
example_interfaces__srv__AddTwoInts_Request req;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){while(1){};}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void service_callback(const void * req, void * res){
  example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
  example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

  //printf("Service request value: %d + %d.\n", (int) req_in->a, (int) req_in->b);

  res_in->sum = req_in->a + req_in->b;
}

void setup() {
  //set_microros_transports();
  Serial.begin(115200);
  set_microros_serial_transports(Serial);
  delay(1000); 

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "add_twoints_client_rclc", "", &support));

  // create service
  RCCHECK(rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, service_callback));
}

void loop() {
  delay(100);
[... 128 lines cut ...]
main.cpp:(.text.setup+0x9a): undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1
================================================================================ [FAILED] Took 5.24 seconds ================================================================================
```

And my platformio.ini file:```

[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_transport = serial

lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio

```

And the workflow as per the manual:
```
pio lib install # Install dependencies
pio run # Build the firmware
pio run --target upload # Flash the firmware
```

Acuadros95, 2023-07-31:
Closing as macOS is not supported.

@usaotearoa Open a new issue if you still need help.

closed 2023-07-31 by Acuadros95

jkaflik, 2023-08-30:
Sorry to see this issue was closed unresolved. I submitted a PR that fixes issues on MacOS: https://github.com/micro-ROS/micro_ros_platformio/pull/116

pablogs9, 2023-09-04:
Thanks @jkaflik, I'm taking a look at your PR