micro-ROS/micro_ros_platformio #68: undefined reference to `rosidl_typesupport_c...

usaotearoa, 2022-10-08:
I get an "undefined reference to" error when attempting to compile the source code below.

`undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'`

```
Ubuntu 22.04.1 with ROS2 Humble install (desktop)
Visual Studio Code Version: 1.71.2
PlatformIO Core 6.1.4·Home 3.4.3
Platform Teensy 4.1 and micro_ROS
```

To the source code. This is my attempt of porting the `micro-ros_addtwoints_service` example from `micro_ros_arduino` repo.
As far as I can tell, I have all libraries properly installed and my include paths defined.

The problem seems to come from the library difference. In the example `#include <micro_ros_arduino.h>` was used and due to porting it to the teensy and platformio it has been changed to `#include <micro_ros_platformio.h>`.
However, digging deeper into the library I can't see where the problem is.

Source code:
```//#include <micro_ros_arduino.h>
#include <Arduino.h>
#include <micro_ros_platformio.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t service;
rcl_wait_set_t wait_set;

example_interfaces__srv__AddTwoInts_Response res;
example_interfaces__srv__AddTwoInts_Request req;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){while(1){};}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void service_callback(const void * req, void * res){
  example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
  example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

  //printf("Service request value: %d + %d.\n", (int) req_in->a, (int) req_in->b);

  res_in->sum = req_in->a + req_in->b;
}

void setup() {
  //set_microros_transports();
  Serial.begin(115200);
  set_microros_serial_transports(Serial);
  delay(1000); 

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "add_twoints_client_rclc", "", &support));

  // create service
  RCCHECK(rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, service_callback));
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}```

This is the entire compiler output I'm getting:

```
~/Documents/gitclones/AutoBot_uros$ pio run
Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/usr/bin/python3 -m pip install pyyaml
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (5.4.1)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/usr/bin/python3 -m pip install markupsafe==2.0.1
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: markupsafe==2.0.1 in /usr/lib/python3/dist-packages (2.0.1)
Configuring teensy41 with transport serial
micro-ROS already built
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.22cf9b6
Building in release mode
Linking .pio/build/teensy41/firmware.elf
.pio/build/teensy41/src/main.cpp.o: In function `setup':
main.cpp:(.text.setup+0x9a): undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1
========================================================== [FAILED] Took 3.82 seconds ==========================================================
```

pablogs9, 2022-10-10:
`example_interfaces` is not build by default, add it to the repo using [extra packages](https://github.com/micro-ROS/micro_ros_platformio#extra-packages)

usaotearoa, 2022-10-15:
@pablogs9 
Please confirm:
Am I correct in assuming that the below is the correct workflow for adding example_interfaces?

1. Navigate to workspace/extra_packages (in my case: "~/Documents/gitclones/AutoBot_uros/extra_packages")
2. Then simply: git clone https://github.com/ros2/example_interfaces.git

Or am I missing something?

usaotearoa, 2022-10-15:
I have tried both approaches above. However, I can't get it to properly build (step "pio run"). I did "clean" before and delete the ".pio" directory to make sure it completely rebuilds.
But it keeps throwing this alarm:
```.pio/build/teensy41/src/main.cpp.o: In function `setup':
main.cpp:(.text.setup+0x54): undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1```

There were some path issues in the .h files (add_two_ints__struct.h, add_two_ints__type_support.h and others, which I have solved, as can be seen below)

```#include "./detail/add_two_ints__struct.h"
#include "./detail/add_two_ints__functions.h"
//"example_interfaces/srv/detail/add_two_ints__type_support.h"
#include "./detail/add_two_ints__type_support.h"```

This is what the directory structure looks like. 

```administrator@dev01:~/Documents/gitclones/AutoBot_uros$ tree -T 5
.
├── extra_packages
│   └── example_interfaces
│       ├── action
│       │   ├── detail
│       │   │   ├── fibonacci__functions.h
│       │   │   ├── fibonacci__rosidl_typesupport_introspection_c.h
│       │   │   ├── fibonacci__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── fibonacci__struct.h
│       │   │   └── fibonacci__type_support.h
│       │   └── fibonacci.h
│       ├── msg
│       │   ├── bool.h
│       │   ├── byte.h
│       │   ├── byte_multi_array.h
│       │   ├── char.h
│       │   ├── detail
│       │   │   ├── bool__functions.h
│       │   │   ├── bool__rosidl_typesupport_introspection_c.h
│       │   │   ├── bool__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── bool__struct.h
│       │   │   ├── bool__type_support.h
│       │   │   ├── byte__functions.h
│       │   │   ├── byte_multi_array__functions.h
│       │   │   ├── byte_multi_array__rosidl_typesupport_introspection_c.h
│       │   │   ├── byte_multi_array__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── byte_multi_array__struct.h
│       │   │   ├── byte_multi_array__type_support.h
│       │   │   ├── byte__rosidl_typesupport_introspection_c.h
│       │   │   ├── byte__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── byte__struct.h
│       │   │   ├── byte__type_support.h
│       │   │   ├── char__functions.h
│       │   │   ├── char__rosidl_typesupport_introspection_c.h
│       │   │   ├── char__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── char__struct.h
│       │   │   ├── char__type_support.h
│       │   │   ├── empty__functions.h
│       │   │   ├── empty__rosidl_typesupport_introspection_c.h
│       │   │   ├── empty__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── empty__struct.h
│       │   │   ├── empty__type_support.h
│       │   │   ├── float32__functions.h
│       │   │   ├── float32_multi_array__functions.h
│       │   │   ├── float32_multi_array__rosidl_typesupport_introspection_c.h
│       │   │   ├── float32_multi_array__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── float32_multi_array__struct.h
│       │   │   ├── float32_multi_array__type_support.h
│       │   │   ├── float32__rosidl_typesupport_introspection_c.h
│       │   │   ├── float32__rosidl_typesupport_microxrcedds_c.h
│       │   │   ├── float32__struct.h
│       │   │   ├── float32__type_support.h
│       │   │   ├── float64__functions.h
[... 150 lines cut ...]
│           │   ├── trigger__functions.h
│           │   ├── trigger__rosidl_typesupport_introspection_c.h
│           │   ├── trigger__rosidl_typesupport_microxrcedds_c.h
│           │   ├── trigger__struct.h
│           │   └── trigger__type_support.h
│           ├── set_bool.h
│           └── trigger.h
├── include
│   └── README
├── lib
│   └── README
├── platformio.ini
├── src
│   └── main.cpp
└── test
    └── README

12 directories, 206 files```

And of course it wouldn't be complete without the entire source code (this is a link to a github repo that is mirrored and public)
https://github.com/usaotearoa/public_troubleshooting.git

Please let me know what I'm missing. Once I have the simple service node working I will also create a pull request and add it to examples.

Thanks!

closed 2022-10-21 by usaotearoa