micro-ROS/micro_ros_platformio #71: Meta configuration variables; where is an overview over possible values

Gleis1234, 2022-10-11:
Question to the Documentation
I need to configure the client transport protocol for an arduino client. In meta files I found configuration variables, but I do not find a list of possible values for the variables.
Can someone give advice how to set variables for serial transport?

closed 2022-10-11 by Gleis1234

Acuadros95, 2022-10-13:
Hi @Gleis1234, documentation on available parameters can be found [here](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration).

Anyway, you don't need to modify this file to use or configure the serial transport. 
If you need help, detail your use case or your configuration needs and we will give you advice on how to proceed.

Gleis1234, 2022-10-13:
Hi @Acuadros95 thank you for your answer. I got my config to run by adding ```board_microros_transport = serial``` to platformio.ini
and init the serial port with:
```
Serial.begin(11520);
set_microros_serial_transports(Serial);
```
I was a bit confused by the lot of config parameters and could not find a clear description of their meaning.
Thank you a lot.