micro-ROS/micro_ros_platformio #75: unable to include extra_packages

andrew-cra2019, 2022-10-19:
## Issue template

- Hardware description: Teensy 4.1
- Installation type: Ubuntu 22.04.1, ROS2 (humble)
- Version or commit hash: ROS-humble
- Platformio Core 6.1.4·Home 3.4.3 withing VS-Code 1.72.2

#### Steps to reproduce the issue
1. Take Arduino -> Examples -> INCOMPATIBLE -> micro_ros_arduino -> "micro-ros_addtwoints_service"
2. Create project in platformio
3. set-up your platformio.ini as the below code-sample 
4. make sure the #include section matches source code below. There is also a separate code-snippet
5. in setup, make sure to add the transport protocol. Line should look like this "set_microros_serial_transports(Serial);"
6. copy the "example_interfaces" directory from Arduino library into "extra_packages" folder. (compare dir-structure below)
7. use proper build workflow: (a) clean, or delete .pio folder, b) pio lib install, c) pio run, d) pio run 

#### Expected behavior
Compile without error

#### Actual behavior
fails at step 7 c) with compiler error, see below for details

#### Additional information
Folder to copy into "extra_packages"

`/Arduino/libraries/micro_ros_arduino-2.0.5-humble/src/example_interfaces`

Compiler error

`undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'`

platfomio.ini

```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_transport = serial
monitor_port = /dev/ttyUSB0
board_microros_distro = humble

lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```

Include section

```
#include <Arduino.h>
#include <Wire.h>
#include <micro_ros_platformio.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>
```

Source code
```
#include <Arduino.h>
#include <micro_ros_platformio.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t service;
rcl_wait_set_t wait_set;

example_interfaces__srv__AddTwoInts_Response res;
example_interfaces__srv__AddTwoInts_Request req;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){while(1){};}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void service_callback(const void * req, void * res){
  example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
  example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

  //printf("Service request value: %d + %d.\n", (int) req_in->a, (int) req_in->b);

  res_in->sum = req_in->a + req_in->b;
}

void setup() {
  set_microros_serial_transports(Serial);  
  Serial.begin(115200);
  delay(1000); 

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "add_twoints_client_rclc", "", &support));

  // create service
  RCCHECK(rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, service_callback));
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```

Directory structure

```
extra_packages$ tree -L 3
.
└── example_interfaces
    ├── action
    │   ├── detail
    │   └── fibonacci.h
    ├── msg
    │   ├── bool.h
    │   ├── byte.h
    │   ├── byte_multi_array.h
    │   ├── char.h
    │   ├── detail
    │   ├── empty.h
    │   ├── float32.h
    │   ├── float32_multi_array.h
    │   ├── float64.h
    │   ├── float64_multi_array.h
    │   ├── int16.h
    │   ├── int16_multi_array.h
    │   ├── int32.h
    │   ├── int32_multi_array.h
    │   ├── int64.h
    │   ├── int64_multi_array.h
    │   ├── int8.h
    │   ├── int8_multi_array.h
    │   ├── multi_array_dimension.h
    │   ├── multi_array_layout.h
    │   ├── rosidl_generator_c__visibility_control.h
    │   ├── rosidl_typesupport_introspection_c__visibility_control.h
    │   ├── rosidl_typesupport_microxrcedds_c__visibility_control.h
    │   ├── string.h
    │   ├── u_int16.h
    │   ├── u_int16_multi_array.h
    │   ├── u_int32.h
    │   ├── u_int32_multi_array.h
    │   ├── u_int64.h
    │   ├── u_int64_multi_array.h
    │   ├── u_int8.h
    │   ├── u_int8_multi_array.h
    │   └── w_string.h
    └── srv
        ├── add_two_ints.h
        ├── detail
        ├── set_bool.h
        └── trigger.h

7 directories, 36 files
```

Acuadros95, 2022-10-19:
You should clone the complete repository in the `extra_packages` folder, not the sources: https://github.com/ros2/example_interfaces

If the `package.xml` file is missing, the build system will not detect the package.

andrew-cra2019, 2022-10-19:
I have tried that as well, but it had the same result.
Is it possible I'm missing a step somehow including something?

Acuadros95, 2022-10-19:
What is your output of the micro-ROS build, does the package appear on the download steps?

Does the compilation error happen if you clean and rebuild micro-ROS?

andrew-cra2019, 2022-10-19:
So do I need to remove the
#include <example_interfaces/srv/add_two_ints.h>
An dit should be added automatically by compiling?

Acuadros95, 2022-10-19:
No, that header will be generated on compilation time from the repository source code

andrew-cra2019, 2022-10-19:
It still failed, when trying to compile, but it definetiley looks different. OMG.
I feel it's really close to being solved! 

Here are the outputs from each step, after I have removed the 
`#include <example_interfaces/srv/add_two_ints.h>`

```pio lib install

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg install` instead.

Resolving teensy41 dependencies...
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio
git version 2.34.1
Cloning into '/home/administrator/.platformio/.cache/tmp/pkg-installing-kam9kcf_'...
remote: Enumerating objects: 67, done.
remote: Counting objects: 100% (67/67), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 67 (delta 10), reused 44 (delta 4), pack-reused 0
Receiving objects: 100% (67/67), 125.47 KiB | 980.00 KiB/s, done.
Resolving deltas: 100% (10/10), done.
Library Manager: micro_ros_platformio@0.0.1+sha.22cf9b6 has been installed!```

And the next step:

```pio run
Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/usr/bin/python3 -m pip install pyyaml
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (5.4.1)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/usr/bin/python3 -m pip install markupsafe==2.0.1
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: markupsafe==2.0.1 in /usr/lib/python3/dist-packages (2.0.1)
Configuring teensy41 with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Downloading micro-ROS library
         - Downloaded microcdr
         - Downloaded microxrcedds_client
         - Downloaded rcl_action
         - Downloaded rcl
         - Downloaded rcl_lifecycle
         - Downloaded rcl_yaml_param_parser (ignored)
         - Downloaded rclc
         - Downloaded rclc_parameter
         - Downloaded rclc_examples (ignored)
         - Downloaded rclc_lifecycle
         - Downloaded micro_ros_utilities
         - Downloaded rcutils
         - Downloaded micro_ros_msgs
[... 366 lines cut ...]
CMake Warning at /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_typesupport_introspection_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_typesupport_introspection_c/cmake/rosidl_typesupport_introspection_cConfig.cmake:41 (include)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/administrator/Documents/gitclones/AutoBot_uros/.pio/libdeps/teensy41/micro_ros_platformio/build/mcu/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

[... cut ...]
  'action_msgs' in your package.xml.
Call Stack (most recent call first):
  CMakeLists.txt:16 (rosidl_generate_interfaces)

---
Failed   <<< example_interfaces [1.80s, exited with code 1]

================================================================================================================================================= [FAILED] Took 84.74 seconds =================================================================================================================================================```

andrew-cra2019, 2022-10-19:
This is my package.xml

```<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>example_interfaces</name>
  <version>0.10.1</version>
  <description>Contains message and service definitions used by the examples.</description>

  <maintainer email="mabel@openrobotics.org">Mabel Zhang</maintainer>

  <license>Apache License 2.0</license>

  <author email="dthomas@openrobotics.org">Dirk Thomas</author>
  <author email="jacob@openrobotics.org">Jacob Perron</author>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <buildtool_depend>rosidl_default_generators</buildtool_depend>

  <exec_depend>rosidl_default_runtime</exec_depend>

  <member_of_group>rosidl_interface_packages</member_of_group>

  <export>
    <build_type>ament_cmake</build_type>
    <ros1_bridge mapping_rules="mapping_rules.yaml"/>
  </export>
</package>```

andrew-cra2019, 2022-10-19:
OMG, You are amazing!
Thank you for your generous help! How can I show my appreciation?
I want to give back, contribute, donate... Let me know, saved my upcoming 4-day weekend ! :-)

```teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:78760, data:10136, headers:8380   free for files:8029188
teensy_size:    RAM1: variables:30464, code:76088, padding:22216   free for local variables:395520
teensy_size:    RAM2: variables:12384  free for malloc/new:511904
Building .pio/build/teensy41/firmware.hex
================================================================================================================================================ [SUCCESS] Took 163.83 seconds ```

<CLOSED>

closed 2022-10-19 by andrew-cra2019

Acuadros95, 2022-10-20:
Glad to help :partying_face:!

We are always open to contributions on our repositories. If you are working on something cool please reach out in slack or by email to have a little chat about it.