micro-ROS/micro_ros_platformio #79: sensor_msgs/Range support structure fails checks when compiling

andrew-cra2019, 2022-10-29:
## Issue template
```
Hardware description: Teensy 4.1
Installation type: Ubuntu 22.04.1, ROS2 (humble)
Version or commit hash: ROS-humble
Platformio Core 6.1.4·Home 3.4.3 withing VS-Code 1.72.2
```

#### Steps to reproduce the issue

Take Source code below, attempt to compile
Create project in platformio
set-up your platformio.ini as the below code-sample
use proper build workflow: (a) clean, and delete .pio folder, b) pio lib install, c) pio run,

#### Expected behavior
Compile without error 

#### Actual behavior
Causes an error:
```
src/main.cpp: In function 'void setup()':
.pio/libdeps/teensy41/micro_ros_platformio/libmicroros/include/rosidl_runtime_c/message_type_support_struct.h:78:59: error: 'rosidl_typesupport_c__get_message_type_support_handle__sesnsor_msgs__msg__Range' was not declared in this scope
     rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
                                                           ^
src/main.cpp:107:43: note: in definition of macro 'RCCHECK'
 #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
                                           ^
src/main.cpp:330:64: note: in expansion of macro 'ROSIDL_GET_MSG_TYPE_SUPPORT'
   RCCHECK(rclc_publisher_init_default(&publisher_range, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(sesnsor_msgs, msg, Range), "micro_ros_platformio_node_publisher_range"));
```

This is in line causing the error
` RCCHECK(rclc_publisher_init_default(&publisher_range, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(sesnsor_msgs, msg, Range), "micro_ros_platformio_node_publisher_range"));`

#### Additional information
directory tree of .pio clearly shows the presence of sensor_msgs and the supporting structure
```
.pio
├── bin
└── libdeps
    └── teensy41
        └── micro_ros_platformio
            ├── libmicroros
            │   ├── include
            │   │   ├── sensor_msgs
            │   │   │   ├── msg
            │   │   │   │   ├── battery_state.h
            │   │   │   │   ├── camera_info.h
            │   │   │   │   ├── channel_float32.h
            │   │   │   │   ├── compressed_image.h
            │   │   │   │   ├── detail
            │   │   │   │   │   ├── battery_state__functions.c
            │   │   │   │   │   ├── battery_state__functions.h
            │   │   │   │   │   ├── ...
            │   │   │   │   │   ├── range__functions.c
            │   │   │   │   │   ├── range__functions.h
            │   │   │   │   │   ├── range__rosidl_typesupport_introspection_c.h
            │   │   │   │   │   ├── range__rosidl_typesupport_microxrcedds_c.h
            │   │   │   │   │   ├── range__struct.h
            │   │   │   │   │   ├── range__type_support.c
            │   │   │   │   │   ├── range__type_support.h
            │   │   │   │   │   ├── region_of_interest__functions.c
            │   │   │   │   │   ├── region_of_interest__functions.h
            │   │   │   │   │   ├── ...
            │   │   │   │   ├── fluid_pressure.h
            │   │   │   │   ├── ...
            │   │   │   │   ├── range.h
            │   │   │   │   ├── region_of_interest.h
            │   │   │   │   ├── ...
[... 436 lines cut ...]
// 5) Strength_L (low 8bit)
// 6) Strength_H (high 8bit)
// 7) Reserved bytes
// 8) Original signal quality degree
// 9) Checksum parity bit (low 8bit), Checksum = Byte1 + Byte2 +…+Byte8. This is only a low 8bit though
while(TFMiniPlus.available()>=9) // When at least 9 bytes of data available (expected number of bytes for 1 signal), then read
{
  // debugln("Check2");
  if((0x59 == TFMiniPlus.read()) && (0x59 == TFMiniPlus.read())) // byte 1 and byte 2
  {
    unsigned int t1 = TFMiniPlus.read(); // byte 3 = Dist_L
    unsigned int t2 = TFMiniPlus.read(); // byte 4 = Dist_H
    t2 <<= 8;
    t2 += t1;
    TFMiniVal = t2*10;
    t1 = TFMiniPlus.read(); // byte 5 = Strength_L
    t2 = TFMiniPlus.read(); // byte 6 = Strength_H
    t2 <<= 8;
    t2 += t1;
    TFMiniStrength = t2;
    for(int i=0; i<3; i++)TFMiniPlus.read(); // byte 7, 8, 9 are ignored
  }
}
}
```

pablogs9, 2022-11-02:
ROSIDL_GET_MSG_TYPE_SUPPORT(**sesnsor_msgs**, msg, Range), "micro_ros_platformio_node_publisher_range")

pablogs9, 2022-11-02:
Please read the compiler errors...

closed 2022-11-02 by pablogs9