micro-ROS/micro_ros_platformio #82: Using more than 4 timers ends up in an error loop.

Yaccobv2, 2022-11-08:
Hello,
Recently I tried to use multiple publishers controlled by multiple timers, but it appears that using more than 4 timers causes an error and the program ends up in an error loop. I would also like to mention that according to Teensy 4.0 documentation (https://www.pjrc.com/store/teensy40.htmla), it has 4 Interval Timers. Is there any way to use more than 4 micro-ros timers in my setup?

- Hardware description: teensy 4.0
- RTOS: Teensy with Arduino
- Installation type: micro_ros_platformio
- Version or commit hash: galactic

clcon.meta:
 "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=2",
                "-DRMW_UXRCE_MAX_PUBLISHERS=6",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=6",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=2",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]

Acuadros95, 2022-11-08:
Hi @Yaccobv2,

Each timer uses a guard condition internally to handle its trigger status. We will take care to add this to the documentation and add another option to configure the number of timers in a more clear way.

On the meantime, you can increase the `RMW_UXRCE_MAX_GUARD_CONDITION` value in your colcon.meta, which defaults to 4.

Yaccobv2, 2022-11-08:
Thanks, it works!

closed 2022-11-08 by Yaccobv2