micro-ROS/micro_ros_platformio #85: Esp32 micro ros transport wifi

Ben01233210, 2022-11-21:
## Issue template
I am trying to connect my esp32 over wifi with the micro ros agent. It runs in a docker container. My machine is a Ubuntu 22.04. 

- Hardware description: esp32-wroom-32 (Az Delivery)
- RTOS: no Rtos
- Installation type: platform io
- Version or commit hash: humble

#### Steps to reproduce the issue
My code is working with serial transport, i tried also with the example code form adruino and platform io adapted for wifi, nothing woked.

But here my code for the wifi:
```
IPAddress agent_ip(192, 168, 1, 100);
size_t agent_port = 8888;

char ssid[] = "ssid";
char psk[] = "pswd";
set_microros_wifi_transports(ssid, psk, agent_ip, agent_port);

```
My platform.ini:
 ```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
board_microros_transport = wifi
monitor_filters = 
	direct
	esp32_exception_decoder
lib_deps = 
	waspinator/AccelStepper@^1.61
	https://github.com/micro-ROS/micro_ros_platformio
```
Micro ros agent in docker (it runs): 
`docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6`

#### Expected behavior
working (connecting to agent) :)
#### Actual behavior
No output on docker. I tried with serial print the program runs and I cloud print all the wifi credentials 

#### Additional information
Esp is in wifi range and i tried some time before with esp idf and it worked also. 
Thanks for help and super cool project thanks for making it possible!

pablogs9, 2022-11-22:
Share code for replicating this issue. 

Which transport are you using?
Does the board connect to wifi with your setup without micro-ROS?

Ben01233210, 2022-11-22:
I am trying to use wifi as transport for micro ros but not getting it to work, with serial transport it worked.

Before I used micro ros with platform io, i used esp idf. There the wifi connection worked with micro ros and also the agent worked so there were no problems.

I don´t know how i should test the wifi connection without using micro ros.
Thanks for the response! 

```
#include <Arduino.h>

#include <micro_ros_platformio.h>

#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

IPAddress agent_ip(192, 168, 178, 88);
size_t agent_port = 8888;

char ssid[] = "ssid";
char psk[] = "pwsd";  // change to correct ssid and pswd in my code

rcl_subscription_t subscriber;
std_msgs__msg__Int32 msg;

rclc_executor_t exe;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

void spin_motor(const void *msgin)
{
	const std_msgs__msg__Int32 *msg = (const std_msgs__msg__Int32 *)msgin;
	digitalWrite(GPIO_NUM_17, HIGH);
	delay(msg->data * 100);
	digitalWrite(GPIO_NUM_17, LOW);
}

void setup()
{
	set_microros_wifi_transports(ssid, psk, agent_ip, agent_port);
	pinMode(GPIO_NUM_17, OUTPUT);
	digitalWrite(GPIO_NUM_17, HIGH);

	digitalWrite(GPIO_NUM_17, HIGH);
	delay(GPIO_NUM_17 * 100);
	digitalWrite(GPIO_NUM_17, LOW);

	rcl_allocator_t allocator = rcl_get_default_allocator();

	rclc_support_init(&support, 0, NULL, &allocator);

	// Init default node
	rclc_node_init_default(&node, "motor_driver", "", &support);

	rclc_subscription_init_default(
		&subscriber, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "motor_pos");

	rclc_executor_init(&exe, &support.context, 1, &allocator);

	rcl_ret_t rc = rclc_executor_add_subscription(
		&exe, &subscriber, &msg,
		&spin_motor, ON_NEW_DATA);
}

void loop()
{
	rclc_executor_spin_some(&exe, RCL_MS_TO_NS(100));
}
```

pablogs9, 2022-11-23:
- Does your network works without micro-ROS?
- Could you link here the transport implementation you are using?

pablogs9, 2022-11-24:
> I am sorry I don't now what you mean with your second point.

In general, I just need instructions for replicating your issue. Share here your project so we can build and reproduce your problem.

Ben01233210, 2022-11-25:
Hi, 
sorry for replying so late. I switched my location and  also the wifi. I used the same code like before but now it works. Probably it was something with the wifi. The only difference I know of is that I now saved it in the default location of platfrom io and not in a costume one like I did before. In about a week I am back in the other location. I`ll see if I can set it up then properly.
But thank you very much!

closed 2022-11-26 by pablogs9