micro-ROS/micro_ros_platformio #88: ESP32 setup hangs at rclc_support_init_with_options()

dixi83, 2022-12-16:
## Issue template

- Hardware description: ESP32 (Wemos d1 R32)
- RTOS: ?? not shure... 
- Installation type: platformio
- Version or commit hash: humble

#### Steps to reproduce the issue
Hi there,

After successfully implemented Micro ROS on my Teensy 4.1 I gave it a try on a ESP32 but for some reason it it hangs on rclc_support_init_with_options() after setting my domain ID.

here is my code:
```
#include <Arduino.h>
#include <micro_ros_platformio.h>

#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/bool.h>

#define ROS_DOMAIN_ID 10
#define WIFI_SSID "someSSID"
#define WIFI_PASS "somePW"

rclc_executor_t         executor;
rclc_support_t          support;
rcl_allocator_t         allocator;
rcl_node_t              node;
rcl_timer_t             timer;

rcl_publisher_t                     publisher;
std_msgs__msg__Float32MultiArray    msg_out_act_position;

rcl_subscription_t                  subscriber_position;
std_msgs__msg__Bool                 msg_in_new_position;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop(temp_rc);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(rcl_ret_t returnCode){
    Serial.print("Error in ROS with return code: ");
    Serial.println(returnCode);
    while(1) {
        digitalWrite(LED_PIN, !digitalRead(LED_PIN));
        delay(500);
    }
}

void subscription_cb_new_position(const void * msgIn) {
    Serial.print("Got a ros 'msg_in_new_position' message!");
}

void setup() {
    Serial.begin(115200);
    while (!Serial);

    IPAddress agent_ip(172, 10, 10, 182);
    size_t agent_port = 8888;

    char SSID[] = WIFI_SSID;
    char PASS[] = WIFI_PASS;

    set_microros_wifi_transports(SSID, PASS, agent_ip, agent_port);
    Serial.println("01: Initialized Wifi");

    delay(1000);

    allocator = rcl_get_default_allocator();
    Serial.println("02: Initialized ROS allocator");

    rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
    Serial.println("03: Initialized ROS init options");
    RCCHECK(rcl_init_options_init(&init_options, allocator));
    Serial.println("04: processed ROS init options");
    RCCHECK(rcl_init_options_set_domain_id(&init_options, 10));
    Serial.println("05: set ROS domain ID");
    RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));
    Serial.println("06: Initialize ROS support with options");

    rcl_ret_t returnNode = rclc_node_init_default(&node, "dixy_transporter", "", &support);
    if(returnNode != 0) {
        error_loop(returnNode);
    }
    Serial.println("07: Initialized ROS returnNode");

    // create publisher "act_position"
    RCCHECK(
        rclc_publisher_init_best_effort(
            &publisher,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray),
            "act_position"
        )
    );
    Serial.println("08: Initialized ROS publisher");

    // create subscriber "new_position"
    RCCHECK(
        rclc_subscription_init_best_effort(
            &subscriber_position,
            &node,
            ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray),
            "new_position"
        )
    );
    Serial.println("09: Initialized ROS subscriber_position");

    // create executor
    RCCHECK(
        rclc_executor_init(
            &executor, 
            &support.context, 
            1, 
            &allocator
        )
    );
    Serial.println("10: Initialized ROS executor");

    RCCHECK(
        rclc_executor_add_subscription(
            &executor, 
            &subscriber_position, 
            &msg_in_new_position, 
            &subscription_cb_new_position, 
            ON_NEW_DATA
        )
    );
    Serial.println("11: Initialized ROS subscribers with the executor");
    Serial.println("-----------------end setup--------------------");
}
//*
```
* I skipped the loop because this is not relevant at the moment

Any idea what is going wrong here, almost the same code is running without issues on my teensy 4.1 using native_ethernet

Thanks, MartijnD

#### Expected behavior
Continue after rclc_support_init_with_options

#### Actual behavior
This is the serial out put:
```
01: Initialized Wifi
02: Initialized ROS allocator
03: Initialized ROS init options
04: processed ROS init options
05: set ROS domain ID
Error in ROS with return code: 1
```

#### Additional information
My platformio.ini:
```
[env:wemos_d1_uno32]
platform = espressif32
board = wemos_d1_uno32
framework = arduino
board_microros_transport = wifi
monitor_speed = 115200
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```

pablogs9, 2022-12-19:
Does your micro-ROS Agent print any log using`-v6`?

dixi83, 2022-12-25:
Hello Pablo, 

Thanks for your reply,
It seemed to be a network issue (I switched test environment...) I am sorry I did not realized this earlier.

Martijn

closed 2022-12-25 by dixi83