micro-ROS/micro_ros_platformio #90: incluing custom message is not found/compiled by micro_ros

dixi83, 2022-12-29:
- Hardware description: ESP32 (wemos D1 R32)
- RTOS: ?
- Installation type: PlatformIO
- Version or commit hash: humble

### Steps to reproduce the issue
- `cd my_ws/src`
- `ros2 pkg create --build-type ament_cmake custom_messages`
- `mkdir msg`
- add the file `Position2D.msg` and add the following content:
```
float32 x
float32 y
```
- copied most of the content from [here](https://github.com/micro-ROS/micro_ros_platformio/tree/main/ci/extra_packages/my_custom_message) to my `CMakeLists.txt` and `package.xml`
- changed the name of the msg file in my CMakeLists file to match my msg file
- `colcon build` (no errors)
- copied my package directory and files to my platformio project in (my_project)/extra_packages/
[image]
- `platformio run --target clean `
```
Processing wemos_d1_uno32 (platform: espressif32; board: wemos_d1_uno32; framework: arduino)
--------------------------------------------------------------------------------------------------------
Removing .pio/build/wemos_d1_uno32
Done cleaning
===================================== [SUCCESS] Took 0.31 seconds =====================================
 *  Terminal will be reused by tasks, press any key to close it.
```
- `platformio run`
```
Processing wemos_d1_uno32 (platform: espressif32; board: wemos_d1_uno32; framework: arduino)
--------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_uno32.html
PLATFORM: Espressif 32 (5.2.0) > WEMOS D1 R32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/martijn/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/martijn/.platformio/penv/lib/python3.10/site-packages (6.0)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/martijn/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/martijn/.platformio/penv/lib/python3.10/site-packages (2.0.1)
Configuring wemos_d1_uno32 with transport wifi
micro-ROS already built
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.22cf9b6
|   |-- WiFi @ 2.0.0
|-- MotorDriverD1-R32 @ 1.0.1+sha.724921a
|-- MicrosTimer @ 1.0.1+sha.ccad1d0
|-- MilisTimer @ 1.0.1+sha.f345b4b
|-- PreMo-Virtual Path Following @ 2.1.0+sha.59ee022
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio/build/wemos_d1_uno32/src/ROS.cpp.o
Compiling .pio/build/wemos_d1_uno32/src/functions.cpp.o
Compiling .pio/build/wemos_d1_uno32/src/motors.cpp.o
Compiling .pio/build/wemos_d1_uno32/src/main.cpp.o
Building .pio/build/wemos_d1_uno32/bootloader.bin
Generating partitions .pio/build/wemos_d1_uno32/partitions.bin
esptool.py v4.2.1
Creating esp32 image...
[... 92 lines cut ...]
Building .pio/build/wemos_d1_uno32/firmware.bin
esptool.py v4.2.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
===================================== [SUCCESS] Took 18.02 seconds =====================================
```
- added `#include <custom_messages/msg/Position2D.h>` did a `platformio run` and it fails:
```
fatal error: custom_messages/msg/Position.h: No such file or directory
 #include <custom_messages/msg/Position2D.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Expected behavior
like described [here](@Acuadros95 in https://github.com/micro-ROS/micro_ros_platformio/issues/42#issuecomment-1192364554_) by @Acuadros95

### Actual behavior
```
fatal error: custom_messages/msg/Position.h: No such file or directory
 #include <custom_messages/msg/Position2D.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

dixi83, 2022-12-29:
I think I found my mistake....
apparently `platformio run --target clean` does NOT clean the micro-ROS library. I needed to run `platformio run --target clean_microros` to do a complete rebuild of the library, like mentioned [here](https://github.com/micro-ROS/micro_ros_platformio#how-to-add-to-your-project)

Hope this helps others...

pablogs9, 2022-12-30:
When building from scratch, on a cleaned micro-ROS library, does your message appears in the include folder? Can you paste here the micro-ROS build output log?

dixi83, 2022-12-30:
Hi Pablo,
Thanks for your reply! 
Yesterday evening I did a file search on `position.h` (which was actually a typo... because Linux is case sensitive) and there was my file... in:
`<PlatformioProject>/.pio/libdeps/<MyBoard>/micro_ros_platformio/libmicroros/include/custom_messages/msg/position.h`
>> So the compiler converts the CamelCase filename to a snake_case! 
If I look at the examples you [see](https://github.com/micro-ROS/micro_ros_platformio/blob/main/ci/src/main.cpp#L27) it to but I didn't expected it!
Hope this topic helps others implementing there custom messages.

Thanks for the help and effort you offer here, really appreciated!

closed 2022-12-30 by dixi83