micro-ROS/micro_ros_platformio #92: Build step microROS_agent fails every time

usaotearoa, 2023-01-08:
For weeks now, I have been struggling to get the Jetson Nano Developer Kit (4GB, B01 with 2 MIPI-ports) to work properly. I'm trying to get a docker container with ROS2 running. Ideally that would be Humble, but I guess I will have to get by with Foxy, as Galactic is EOL.

This has been tried on multiple Jetson Nano Developer Kits, as I have two of them. Here are the steps outlined of what has been done:

REGULAR DOCKER IMAGE WORKFLOW
Flash the device with the latest sdkmanager to JetPack 4.6.3. (I have also tried flashing it to 32.7.1, but the steps below still fail. If I opt for DeepStream or not also does not change the outcome.)

After the device is finished, log in and run apt-get update && upgrade (In some trials, this step was omitted and yet the following procedure still failed.)
Clone the GitHub repository (git clone https://github.com/dusty-nv/jetson-containers.git)
Ensure "default-runtime": "nvidia" is entered in the /etc/docker/daemon.json, properly formatted and comma added after the prior entry!
Then run one of the following scripts:

    scripts/docker_run.sh -c dustynv/ros:foxy-ros-base-l4t-r32.7.1
    scripts/docker_run.sh -c dustynv/ros:foxy-pytorch-l4t-r32.7.1

This fails because I'm already on 32.7.3.

Modified the l4t_version.sh to fake the output to show R32.7.1 (Even if this step is not needed due to being on Jetpack 4.6.1 and R32.7.1, as mentioned above, the process fails)

Then attempt to run the docker command from step 5 again, and it actually works, loading the docker.
Following the tutorial for microROS for Teensy
(https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/),
the process fails at the following step:

# Build step
ros2 run micro_ros_setup build_agent.sh

Also, the very first source command to source the environment is wrong. It is:

# Source the ROS 2 installation
source /opt/ros/foxy/setup.bash

But it should be:

source /opt/ros/foxy/install/setup.bash

However, despite this small oversight, the agent build still fails.

I know this is not related to the Jetpack version, as mentioned it fails with 4.6.1 as well. At some point, I was able to get it working, but I'm unsure what I have done differently and why it suddenly wouldn't work anymore. Output logs are to follow below.
BUILD DOCKER IMAGE WORKFLOW

Then I decided to build the docker container from source, as I thought maybe there is something intrinsically different in my system, and building from source will sort all dependencies properly. This has been tried under the following circumstances:

    With the device flashed to JetPack 4.6.1 (R32.7.1) - Directly after flashing (no apt-get update && upgrade)
    With the device flashed to JetPack 4.6.1 (R32.7.1) - With apt-get update && upgrade
    With the device flashed to JetPack 4.6.3 (R32.7.2) and the modified l4t_version.sh to fake the output to show R32.7.1 - Directly after flashing (no apt-get update && upgrade)
    With the device flashed to JetPack 4.6.3 (R32.7.2) and the modified l4t_version.sh to fake the output to show R32.7.1 - With apt-get update && upgrade

No matter what I attempt to do, the build process also fails.

So to summarize, my situation is as follows:

Unable to get the microROS agent to build (ros2 run micro_ros_setup build_agent.sh)
Unable to build the docker image from source

Output logs are to follow below.

Before attempting it to the 12th time, I have tried to make sure the dependencies for fmt are met by:

```
apt-get update
apt-get install -y build-essential
apt-get install libfmt-dev
```

Here the entire output from an attempt on a freshly flashed Jetson Nano with Jetpack 4.6.1 (R32.7.1). entire trace below:
Because I have done this process so many times and am getting tired of typing / copying and pasting all commands, I have thrown them into a shell script. But if I do it manually of with the script, the outcome is the same. Here the script:

#! /bin/bash

[... 330 lines cut ...]
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v6::basic_string_view<char>, spdlog::level::level_enum, fmt::v6::basic_string_view<char>)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::format_error::~format_error()'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::hex_digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_64'
collect2: error: ld returned 1 exit status
make[5]: *** [micro_ros_agent] Error 1
make[4]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< micro_ros_agent [10min 14s, exited with code 2]

Summary: 1 package finished [11min 1s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
root@jetson2:/# 

Would you kindly advice?

pablogs9, 2023-01-09:
We had this error on another issue. Do you remember something @Acuadros95 ?

```
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::signs'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `vtable for fmt::v6::format_error'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::snprintf_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `char fmt::v6::internal::decimal_point_impl<char>(fmt::v6::internal::locale_ref)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_32'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `char fmt::v6::internal::thousands_sep_impl<char>(fmt::v6::internal::locale_ref)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `std::locale fmt::v6::internal::locale_ref::get<std::locale>() const'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `typeinfo for fmt::v6::format_error'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v6::basic_string_view<char>, spdlog::level::level_enum, fmt::v6::basic_string_view<char>)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::format_error::~format_error()'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::hex_digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::digits'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.3.0: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_64'
```

Acuadros95, 2023-01-09:
Yes, check this comment with the solution @usaotearoa : https://github.com/micro-ROS/micro_ros_setup/issues/455#issuecomment-1042635375