micro-ROS/micro_ros_platformio #95: Using ethernet module results in RCL_RET_ERROR being returned when running rclc_support_init

sowmr5, 2023-02-21:
## Using ethernet module results in `RCL_RET_ERROR` being returned when running `rclc_support_init`

- Hardware description: Teensy 4.1
- RTOS: Arduino
- Installation type: PlatformIO example
- Version or commit hash: 1.0.0 (2022-05-25) - judging from changelog in platformio library folder.

#### Steps to reproduce the issue
Run the following code in platformio: https://pastebin.com/p0CZA7zE
When trying to use the ethernet module, it seems to fail when reaching:
`RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));`
The value returned is `RCL_RET_ERROR` but there is no other detail on it. Ethernet itself appears to work as there is an IP on the network which can be pinged.

#### Expected behavior

rclc_support_init returns a non error value and continues running the rest of the program

#### Actual behavior

The teensy will end up in the error_loop function when the RCL_RET_ERROR error is returned in the RCCHECK

pablogs9, 2023-02-21:
Is this code being built?

```c
// Test custom transports
#if defined(MICRO_ROS_TRANSPORT_ARDUINO_CUSTOM)
bool platformio_transport_open(struct uxrCustomTransport * transport) {return false;};
bool platformio_transport_close(struct uxrCustomTransport * transport) {return false;};
size_t platformio_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err) {return 0;};
size_t platformio_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err) {return 0;};
#endif
```

sowmr5, 2023-02-21:
No, it's not being built.

pablogs9, 2023-02-21:
What is the output of your micro-ROS Agent with the flag `-v6`?

sowmr5, 2023-02-21:
Command:

micro-ros-agent udp4 --port 8888 -v6

Output:

[1676993210.570109] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1676993210.570422] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

pablogs9, 2023-02-21:
Your board is not communication with the agent, this is why `rclc_support_init`.

Can you ensure that without micro-ROS your board networking is working?

sowmr5, 2023-02-22:
Thanks for your help. After trying the following demo code:

https://github.com/vjmuzik/NativeEthernet/blob/master/examples/UDPSendReceiveString/UDPSendReceiveString.ino

and sending a UDP packet, the Teensy network appears to be working ok:

Reached end of setupReceived packet of size 2
From 192.168.1.27, port 44744
Contents:
hi

It that test ok, or should a different one be made? As a next step I'll check if there are any issue with the teensy accessing the agent on the host machine.

Thanks again

pablogs9, 2023-02-22:
This test is ok, now you need to ensure that micro-ROS in sending UDP message to port 8888 in your host (where the agent is listening). Probably is a good idea to take a look at send method of the custom transports.