micro-ROS/micro_ros_platformio #96: MICRO_ROS_TRANSPORT_ARDUINO_SERIAL not defined in example

patrickwasp, 2023-02-24:
## Issue template

- Hardware description: Arduino® MKR Zero SAMD21 Cortex-M0+ (mkrzero)
- RTOS: 
- Installation type: `https://github.com/micro-ROS/micro_ros_platformio#1.0.0` using lib_deps
- Version or commit hash: [#1.0.0](https://github.com/micro-ROS/micro_ros_platformio/releases/tag/1.0.0)

#### Steps to reproduce the issue
1. start a new platformio project with the following config
```
[env:mkrzero]
platform = atmelsam
board = mkrzero
framework = arduino
board_microros_transport = serial
board_microros_distro = humble
board_microros_user_meta = config/mkrzero.meta
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio#1.0.0
```

2. copy the example from https://github.com/micro-ROS/micro_ros_platformio/blob/main/examples/micro-ros_publisher/src/Main.cpp
3. build 

#### Expected behavior
successful build

#### Actual behavior
`MICRO_ROS_TRANSPORT_ARDUINO_SERIAL` is undefined so an error is shown that this example only works with arduino and serial transport, but that's what I'm trying to do.

```
#if !defined(MICRO_ROS_TRANSPORT_ARDUINO_SERIAL)
#error This example is only avaliable for Arduino framework with serial transport.
#endif
```

#### Additional information

deleting the check resolves the issue, but that's probably not what I'm supposed to do.

pablogs9, 2023-02-27:
Could you check if using `main` tag instead of `v1.0.0` solves this?

@Acuadros95 if this is solved, we need to plan a new release due to outdated examples

Acuadros95, 2023-02-27:
Hi @patrickwasp,

I guess you are using a platformIO version >= 6.1, which needs this fix for adding definitions properly: https://github.com/micro-ROS/micro_ros_platformio/pull/36

We will generate a new release soon, on the meantime you can use the actual `main` branch for your development.

patrickwasp, 2023-03-01:
Yes @Acuadros95 I'm using 6.1.6. Changing to `main` fixes the issue. I like to work using releases since I've had problems before where changes in the `main` branch of a project can be breaking.

Acuadros95, 2023-03-01:
Closing the issue them, will update when a release is generated.

closed 2023-03-01 by Acuadros95