micro-ROS/micro_ros_platformio #97: PlatformIO connection to ROS2

RRbotics, 2023-03-03:
Hello, I am just a beginner in MicroRos and Ros2: just followed the instructions to use Microros with the PlatformIO extension on VScode. 

I was able to come to the point where I built the project and uploaded it to the microcontroller, however, after that, I can not connect to ROS2. (In the instructions, you have some commands for Docker I guess, but I am using local ROS2 installation). 

Could you please let me know how I can create a ROS2 agent and connect it to a microcontroller? (the way it is done conventionally is you first create and configure firmware, build it, and create a ROS2 agent and built it as well). I am having difficulty with how the process is done with PlatformIO.

pablogs9, 2023-03-03:
Are you opening a micro-ROS Agent? It is not clear for me.

If not, you can:
- Use the dockerized micro-ROS Agent: https://github.com/micro-ROS/micro_ros_platformio#using-the-micro-ros-agent
- Build the micro-ROS Agent package: https://github.com/micro-ROS/micro-ROS-Agent
- Use Vulcanexus distro, which comes with a ready-to-use micro-ROS Agent: https://docs.vulcanexus.org/en/latest/rst/installation/linux_binary_installation.html

RRbotics, 2023-03-03:
I was not able to open a Micro-ROS Agent, but now I realized that I can just use the prebuilt micro-ROS agent. Thanks a lot for your reply. Problem solved.

closed 2023-03-03 by RRbotics