micro-ROS/micro_ros_platformio #98: Basic publisher exemple with teensy36 stops aftern logger setup

FrGe2016, 2023-05-05:
## Issue template

- Hardware description:  Teensy 3.6 plugged to a Raspberry pi 4
- RTOS: 
- Installation type:  basic installation
- Version or commit hash: Humble

#### Steps to reproduce the issue
 platform.ini is modified for the use of a teensy 3.6
          then micro-ros_publisher is built and uploaded without errors. 

#### Expected behavior After ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
                                                should display Running..Logger setup .......Datawriter created
                                                 ros2 topic lists should display a new topic micro_ros_platformio_node_publisher...

 

#### Actual behavior     The display stop after logger setup and nothing else happen

#### Additional information

Acuadros95, 2023-05-16:
There was an error on Teensy builds with serial transport, which was fixed on this PR: https://github.com/micro-ROS/micro_ros_platformio/pull/99

Please clean your project dependencies and give it another try.

FrGe2016, 2023-05-18:
Thanks for your help

I have tried with my old teensy 3.6 and my new teensy 4.0 and had the same result ??????

Acuadros95, 2023-07-31:
Closing as resolved.

For the record, its a good practice to modify your code to check the Agent connection before micro-ROS is initialized. More details on this tutorial: [Reconnections and liveliness](https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html)