micro-ROS/micro_ros_platformio #100: number of timers available

patrickwasp, 2023-05-17:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS?
- Installation type: platformio
- Version or commit hash: https://github.com/micro-ROS/micro_ros_platformio#1.0.0

#### Steps to reproduce the issue
```c++
#include <Arduino.h>
#include <micro_ros_platformio.h>

#define RC_CHECK(fn)             \
  {                              \
    rcl_ret_t temp_rc = fn;      \
    if ((temp_rc != RCL_RET_OK)) \
    {                            \
      rcError();                 \
    }                            \
  }

#define EXECUTE_EVERY_N_MS(MS, X)      \
  do                                   \
  {                                    \
    static volatile int64_t init = -1; \
    if (init == -1)                    \
    {                                  \
      init = uxr_millis();             \
    }                                  \
    if (uxr_millis() - init > MS)      \
    {                                  \
      X;                               \
      init = uxr_millis();             \
    }                                  \
  } while (0)

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>

rclc_support_t g_support;
rcl_node_t g_node;
rclc_executor_t g_executor;
static rcl_allocator_t g_allocator;
rcl_init_options_t g_init_options;

rcl_timer_t g_update_first_timer;
rcl_timer_t g_update_second_timer;
rcl_timer_t g_update_third_timer;
rcl_timer_t g_first_publisher_timer;
rcl_timer_t g_second_publisher_timer;
rcl_timer_t g_third_publisher_timer;

rcl_publisher_t g_first_publisher;
rcl_publisher_t g_second_publisher;
rcl_publisher_t g_third_publisher;

std_msgs__msg__Int32 g_first_msg;
std_msgs__msg__Int32 g_second_msg;
std_msgs__msg__Int32 g_third_msg;

float g_first_value;
float g_second_value;
float g_third_value;

[... 241 lines cut ...]
#### Additional information

If I change the order of the publishers and timers in code then different publishers start working. Is there a built-in limit in `rclc_executor_get_zero_initialized_executor()`, `rcl_get_default_allocator()`, or `rcl_get_zero_initialized_init_options()` that set the number of timers that can be run?

When it comes to timers, does the timeout_ns indicate how often it will run or how long it will wait for the callback to return? If it is just a timeout, how do you set the callback frequency? Are timer callbacks run asynchronously or are they blocking for the rest of the code?

the meta I'm using is

```
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```

Acuadros95, 2023-05-17:
Glad to hear!

> Is 20 a good number, or is it too high?

Only a `RMW_UXRCE_MAX_GUARD_CONDITION` per timer is needed, but you will just waste a bit of memory. 

Closing!

closed 2023-05-17 by Acuadros95

chriswang2001, 2023-05-21:
Here is the complete log

Resolving esp32dev dependencies...
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio.git
git version 2.36.0.windows.1
Cloning into 'C:\Users\haozh\.platformio\.cache\tmp\pkg-installing-4kbgygur'...

Library Manager: micro_ros_platformio@0.0.1+sha.8103c3c has been installed!
Updating metadata for the vscode IDE...
Error: Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
C:\Users\haozh\.platformio\penv\Scripts\python.exe -m pip install pyyaml
Requirement already satisfied: pyyaml in c:\users\haozh\.platformio\penv\lib\site-packages (6.0)

[notice] A new release of pip is available: 23.1 -> 23.1.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Installing markupsafe==2.0.1 with pip at PlatformIO environment
C:\Users\haozh\.platformio\penv\Scripts\python.exe -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in c:\users\haozh\.platformio\penv\lib\site-packages (2.0.1)

[notice] A new release of pip is available: 23.1 -> 23.1.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Configuring esp32dev with transport serial
Downloading micro-ROS dev dependencies
	 - Downloaded ament_cmake
	 - Downloaded ament_lint
	 - Downloaded ament_package
	 - Downloaded googletest
	 - Downloaded ament_cmake_ros
	 - Downloaded ament_index
Building micro-ROS dev dependencies
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 4: invalid start byte:
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\haozh\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProjectDeps()
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 137:
    plb = env.ConfigureProjectLibBuilder()
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1180:
    project.install_dependencies()
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 965:
    if _is_builtin(spec):
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 957:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1093:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 153:
    self.process_extra_options()
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 301:
    self.env.SConscript(
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 157:
    build_microros()
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 99:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 64:
    self.build_dev_environment()
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 107:
    print("Build dev micro-ROS environment failed: \n {}".format(result.stderr.decode("utf-8")))
========================= [FAILED] Took 18.71 seconds =========================