micro-ROS/micro_ros_platformio #101: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 4: invalid start byte when add micro_ros libdeps

chriswang2001, 2023-05-21:
I am using platformio IDE in vscode based on win11. When I am trying to add micro_ros like this in platformio.ini like this

```
lib_deps = 
    https://github.com/micro-ROS/micro_ros_platformio.git
```
 
after saving, the plaform build with error and here is the log

Building micro-ROS dev dependencies
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 4: invalid start byte:
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\haozh\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProjectDeps()
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 137:
    plb = env.ConfigureProjectLibBuilder()
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1180:
    project.install_dependencies()
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 965:
    if _is_builtin(spec):
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 957:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1093:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 153:
    self.process_extra_options()
  File "C:\Users\haozh\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 301:
    self.env.SConscript(
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\haozh\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 157:
    build_microros()
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 99:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 64:
    self.build_dev_environment()
  File "D:\work\micro_ros_test\.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 107:
    print("Build dev micro-ROS environment failed: \n {}".format(result.stderr.decode("utf-8")))

chriswang2001, 2023-05-21:
Can this repo run on Win11? I am sure that platformio IDE for vscode works well on my computer.

chriswang2001, 2023-05-21:
Here is the complete log:

Resolving esp32dev dependencies...
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio.git
git version 2.36.0.windows.1
Cloning into 'C:\Users\haozh.platformio.cache\tmp\pkg-installing-4kbgygur'...

Library Manager: micro_ros_platformio@0.0.1+sha.8103c3c has been installed!
Updating metadata for the vscode IDE...
Error: Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:

framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
tool-esptoolpy @ 1.40501.0 (4.5.1)
toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
C:\Users\haozh.platformio\penv\Scripts\python.exe -m pip install pyyaml
Requirement already satisfied: pyyaml in c:\users\haozh.platformio\penv\lib\site-packages (6.0)
[notice] A new release of pip is available: 23.1 -> 23.1.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Installing markupsafe==2.0.1 with pip at PlatformIO environment
C:\Users\haozh.platformio\penv\Scripts\python.exe -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in c:\users\haozh.platformio\penv\lib\site-packages (2.0.1)

[notice] A new release of pip is available: 23.1 -> 23.1.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Configuring esp32dev with transport serial
Downloading micro-ROS dev dependencies
- Downloaded ament_cmake
- Downloaded ament_lint
- Downloaded ament_package
- Downloaded googletest
- Downloaded ament_cmake_ros
- Downloaded ament_index
Building micro-ROS dev dependencies
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 4: invalid start byte:
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
File "C:\Users\haozh.platformio\platforms\espressif32\builder\main.py", line 312:
target_elf = env.BuildProgram()
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
return self.method(*nargs, **kwargs)
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
env.ProcessProjectDeps()
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
return self.method(*nargs, **kwargs)
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 137:
plb = env.ConfigureProjectLibBuilder()
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
return self.method(*nargs, **kwargs)
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1180:
project.install_dependencies()
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 965:
if _is_builtin(spec):
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 957:
for lb in self.env.GetLibBuilders():
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
return self.method(*nargs, **kwargs)
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1093:
lb = LibBuilderFactory.new(env, lib_dir)
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 60:
obj = globals()[clsname](env, path, verbose=verbose)
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 153:
self.process_extra_options()
File "C:\Users\haozh.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 301:
self.env.SConscript(
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\haozh.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
File "D:\work\micro_ros_test.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 157:
build_microros()
File "D:\work\micro_ros_test.pio\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 99:
builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
File "D:\work\micro_ros_test.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 64:
self.build_dev_environment()
File "D:\work\micro_ros_test.pio\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 107:
print("Build dev micro-ROS environment failed: \n {}".format(result.stderr.decode("utf-8")))
========================= [FAILED] Took 18.71 seconds =========================

Acuadros95, 2023-05-22:
> Can this repo run on Win11?

No, Windows is not currently supported and its not a feature on our roadmap.

closed 2023-05-22 by Acuadros95

chriswang2001, 2023-05-22:
> > Can this repo run on Win11?
> 
> No, Windows is not currently supported and its not a feature on our roadmap.

can you please write this in README, maybe this can save time for people who use platformio on win11 like me