micro-ROS/micro_ros_platformio #103: Unable to create two servers in micro_ros_platformio

beta1238888, 2023-06-07:
## Issue template
I need two servers, but after uploading code to ESP32 and connecting to Agent, the service list shows only one server "/service1"
- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: micro_ros_platformio
- Version or commit hash: humble

#### Steps to reproduce the issue

`#include <Arduino.h>
#include <micro_ros_platformio.h>
#include <stdio.h> 
#include <rcl/rcl.h> 
#include <rcl/error_handling.h> 
#include <rclc/rclc.h> 
#include <rclc/executor.h> 
#include <std_srvs/srv/set_bool.h> 

rcl_service_t service1; 
rcl_service_t service2; 
std_srvs__srv__SetBool_Request req1; 
std_srvs__srv__SetBool_Request req2; 
std_srvs__srv__SetBool_Response res1; 
std_srvs__srv__SetBool_Response res2; 

rclc_executor_t executor; 
rclc_support_t support;
rcl_allocator_t allocator; 
rcl_node_t node; 

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}} 
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}} 

void error_loop(){ 
  while(1){
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    delay(100);
  }
}

void service_callback1(const void * req, void * res){ 
  std_srvs__srv__SetBool_Request * req_in = (std_srvs__srv__SetBool_Request *) req; 
  std_srvs__srv__SetBool_Response * res_in = (std_srvs__srv__SetBool_Response *) res; 

  res_in->success = true; 
  res_in->message.data = (char*) malloc(20 * sizeof(char)); 
  sprintf(res_in->message.data, "Status: %d", req_in->data); 
  res_in->message.size = strlen(res_in->message.data); 

  digitalWrite(LED_BUILTIN, req_in->data); 
}

void service_callback2(const void * req, void * res){ 
  std_srvs__srv__SetBool_Request * req_in = (std_srvs__srv__SetBool_Request *) req; 
  std_srvs__srv__SetBool_Response * res_in = (std_srvs__srv__SetBool_Response *) res; 

  res_in->success = true; 
  res_in->message.data = (char*) malloc(20 * sizeof(char)); 
  sprintf(res_in->message.data, "Status: %d", req_in->data); 
  res_in->message.size = strlen(res_in->message.data); 

  digitalWrite(LED_BUILTIN, !req_in->data); 
}

void setup() {
  Serial.begin(115200); 
  set_microros_serial_transports(Serial);

  pinMode(LED_BUILTIN, OUTPUT); 
  digitalWrite(LED_BUILTIN, HIGH); 
  delay(2000); 

  allocator = rcl_get_default_allocator(); 

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator)); 

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support)); 

  // create first service
  RCCHECK(rclc_service_init_default(
    &service1,
    &node,
    ROSIDL_GET_SRV_TYPE_SUPPORT(std_srvs, srv, SetBool),
    "/service1"));

  // create second service
  RCCHECK(rclc_service_init_default(
    &service2,
    &node,
    ROSIDL_GET_SRV_TYPE_SUPPORT(std_srvs, srv, SetBool),
    "/service2")); 

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator)); 
  RCCHECK(rclc_executor_add_service(&executor, &service1, &req1, &res1, service_callback1)); 
  RCCHECK(rclc_executor_add_service(&executor, &service2, &req2, &res2, service_callback2)); 
}

void loop() {
  delay(100); 
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100))); 
}`
#### Expected behavior
 "/service1" and "/service2" in the service list
#### Actual behavior
Only "/service1", not "/service2"
#### Additional information

pablogs9, 2023-06-07:
Change this: https://github.com/micro-ROS/micro_ros_platformio/blob/8103c3c2bbf52bc2cae5ff95fc61f06dd5792b62/metas/colcon.meta#L8

beta1238888, 2023-06-07:
I changed " -DRMW_UXRCE_MAX_SERVICES=1 " to " -DRMW_UXRCE_MAX_SERVICES=2 " in colcon.meta , then build and upload to ESP32, but still no "/service2"
```
ros@ros-desktop:~$` ros2 service list
/service1
```

pablogs9, 2023-06-08:
You need to change the colcon meta file and build it in a clean environment, in the log you show the micro-ROS library is not being rebuilt.

beta1238888, 2023-06-08:
Thanks for your help, I'm new to microROS and PlatformIO, what is "build it in a clean environment"? I tried doing the following:
After modifying colcon.meta, I clicked "Clean" in the PlatformIO plugin:
[image]
The terminal output:
```
Processing featheresp32 (platform: espressif32; board: featheresp32; framework: arduino)
-------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Removing .pio/build/featheresp32
Done cleaning
================================== [SUCCESS] Took 2.67 seconds ==================================
```
I can see the files removed in pio/build/:
[image]
Then I clicked Build:
[image]
The terminal output:
```
Processing featheresp32 (platform: espressif32; board: featheresp32; framework: arduino)
-------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 (6.3.0) > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/ros/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/ros/.platformio/penv/lib/python3.10/site-packages (6.0)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/ros/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/ros/.platformio/penv/lib/python3.10/site-packages (2.0.1)
Configuring featheresp32 with transport serial
micro-ROS already built
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.8103c3c
Building in release mode
Compiling .pio/build/featheresp32/src/main.cpp.o
Building .pio/build/featheresp32/bootloader.bin
Generating partitions .pio/build/featheresp32/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/featheresp32/libfaa/micro_ros_platformio/platform_code/arduino/clock_gettime.cpp.o
Compiling .pio/build/featheresp32/libfaa/micro_ros_platformio/platform_code/arduino/serial/micro_ros_transport.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/HardwareSerial.cpp.o
Archiving .pio/build/featheresp32/libfaa/libmicro_ros_platformio.a
Indexing .pio/build/featheresp32/libfaa/libmicro_ros_platformio.a
Compiling .pio/build/featheresp32/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/Print.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/USB.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/USBCDC.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/USBMSC.cpp.o
[... 22 lines cut ...]
Compiling .pio/build/featheresp32/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/main.cpp.o
Compiling .pio/build/featheresp32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/featheresp32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/featheresp32/libFrameworkArduino.a
Indexing .pio/build/featheresp32/libFrameworkArduino.a
Linking .pio/build/featheresp32/firmware.elf
Retrieving maximum program size .pio/build/featheresp32/firmware.elf
Checking size .pio/build/featheresp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.0% (used 42716 bytes from 327680 bytes)
Flash: [==        ]  24.8% (used 325185 bytes from 1310720 bytes)
Building .pio/build/featheresp32/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
================================= [SUCCESS] Took 161.49 seconds =================================
```
Then click Upload, after the Agent is connected, there is still only "/service1"

pablogs9, 2023-06-08:
Remove manually `libmicroros` folder. Is this correct @Acuadros95 ?

Acuadros95, 2023-06-08:
Yes, its correct.

You can also do it with a platformIO command: `pio run --target clean_microros  # Clean library`
Or you can try with the `Clean all` option of your vs code extension

beta1238888, 2023-06-08:
Thanks you all, I deleted /libmicroros and rebuilt, it worked, now both servers work. But I have a new problem, here is the code:
```
#include <Arduino.h>
#include <micro_ros_platformio.h>
#include <stdio.h> 
#include <rcl/rcl.h> 
#include <rcl/error_handling.h> 
#include <rclc/rclc.h> 
#include <rclc/executor.h> 
#include <bot_interfaces/srv/led_wink.h> 
#include <bot_interfaces/srv/buzzer_wink.h>

rcl_service_t service1; 
rcl_service_t service2; 

rclc_executor_t executor; 
rclc_support_t support; 
rcl_allocator_t allocator; 
rcl_node_t node; 

bot_interfaces__srv__LedWink_Request req1;
bot_interfaces__srv__LedWink_Response res1;

bot_interfaces__srv__BuzzerWink_Request req2;
bot_interfaces__srv__BuzzerWink_Response res2;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}} 
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}} 

void error_loop(){ 
  while(1){
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    delay(100);
  }
}

void service_callback1(const void * req, void * res){ 
  bot_interfaces__srv__LedWink_Request * req_in = (bot_interfaces__srv__LedWink_Request *) req; 
  bot_interfaces__srv__LedWink_Response * res_in = (bot_interfaces__srv__LedWink_Response *) res; 

  if(req_in->c == 1){
    for(int i = 0;i < 8;i++){
        digitalWrite(2, HIGH);
        delay(1000);
        digitalWrite(2, LOW);
        delay(1000);
    }
  }
  res_in->result = 1;
}

void service_callback2(const void * req, void * res){ 
  bot_interfaces__srv__BuzzerWink_Request * req_in = (bot_interfaces__srv__BuzzerWink_Request *) req; 
  bot_interfaces__srv__BuzzerWink_Response * res_in = (bot_interfaces__srv__BuzzerWink_Response *) res; 

  if(req_in->a == 1){
    for(int i = 0;i < 8;i++){
        digitalWrite(13, HIGH);
        delay(1000);
        digitalWrite(13, LOW);
        delay(1000);
    }
  }
  res_in->result_2 = 1;
}

void setup() {
  Serial.begin(115200); 
  set_microros_serial_transports(Serial);
  
  pinMode(2, OUTPUT);
  pinMode(13, OUTPUT);
  digitalWrite(2, LOW);
  digitalWrite(13, LOW);
  delay(2000); 

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator)); 

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support)); 

  // create first service
  RCCHECK(rclc_service_init_default(
    &service1,
    &node,
    ROSIDL_GET_SRV_TYPE_SUPPORT(bot_interfaces, srv, LedWink),
    "/service1")); 

  // create second service
  RCCHECK(rclc_service_init_default(
    &service2,
    &node,
    ROSIDL_GET_SRV_TYPE_SUPPORT(bot_interfaces, srv, BuzzerWink),
    "/service2")); 

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator)); 
  RCCHECK(rclc_executor_add_service(&executor, &service1, &req1, &res1, service_callback1)); 
  RCCHECK(rclc_executor_add_service(&executor, &service2, &req2, &res2, service_callback2)); 
}

void loop() {
  delay(100); 
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```
When I request /service1 to make the LED blink, and request /service2 in another terminal, /service2 will not start responding until /service1 has finished blinking. Can micro-ROS make all servers respond at the same time?

Acuadros95, 2023-06-08:
> Can micro-ROS make all servers respond at the same time?

You would need a multithread approach to achieve this, which is not available on this repository.

Also, its not a good practice to block inside micro-ROS callbacks, as the service and all the communications will be blocked until the sleeps of that callback are finished.

Closing as the initial issue is solved.

closed 2023-06-08 by Acuadros95

akvavit01, 2024-10-11:
I am encountering the same issue. Manually editing colcon.meta works. However, is there any other solution to this problem? For example, perhaps we can set up something in platformio.ini to change the configuration in colcon.meta.

hippo5329, 2024-10-11:
You may specify user meta file in platformio.ini. Like this,

board_microros_user_meta = user.meta

akvavit01, 2024-10-18:
It works, thanks