micro-ROS/micro_ros_platformio #113: [WIndows10] Lib install errors

Magnum164, 2023-08-13:
## Issue template

Basically found this lib and wanted to try out, clean install no code starting from scratch.

- Hardware description:  Windows 10 VSCode
- RTOS: VSCode w/PLatformIO
- Installation type: Basic

#### Steps to reproduce the issue
Added lib_deps to platformio.ini

#### Expected behavior
lib compile
#### Actual behavior
Fails stating file already exists in build folder, then gives the error "." is invalid.

#### Additional information
Configuring esp32dev with transport serial
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'D:\\ArduinoSketches\\libdeps\\esp32dev\\micro_ros_platformio/build/dev/src':
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\magnu\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 147:
    plb = env.ConfigureProjectLibBuilder()
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1180:
    project.install_dependencies()
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 965:
    if _is_builtin(spec):
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 957:
    for lb in self.env.GetLibBuilders():
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1093:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 153:
    self.process_extra_options()
  File "C:\Users\magnu\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 301:
    self.env.SConscript(
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\magnu\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\ArduinoSketches\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 157:
    build_microros()
  File "D:\ArduinoSketches\libdeps\esp32dev\micro_ros_platformio\extra_script.py", line 99:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "D:\ArduinoSketches\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 63:
    self.download_dev_environment()
  File "D:\ArduinoSketches\libdeps\esp32dev\micro_ros_platformio\microros_utils\library_builder.py", line 94:
    os.makedirs(self.dev_src_folder)
  File "C:\Program Files\Python39\lib\os.py", line 225:
    mkdir(name, mode)

pablogs9, 2023-08-14:
Hello @Magnum164, this package targets a Linux development environment. Is possible for you to test it on a Linux machine? If not, I would recommend you using the [precompiled libraries provided in the micro-ROS for Arduino package](https://github.com/micro-ROS/micro_ros_arduino)

Magnum164, 2023-08-14:
> Hello @Magnum164, this package targets a Linux development environment. Is possible for you to test it on a Linux machine? If not, I would recommend you using the [precompiled libraries provided in the micro-ROS for Arduino package](https://github.com/micro-ROS/micro_ros_arduino)

I have ROS running on a VM. But I don't understand I was literally following the instructions linked to the build on running with vscode. So a little confused if it is not meant for Windows/VScode why is it linking me to install VS code and IDE without redirecting me to the other githubs.

Which I have the other packages as well. Man, I am retired why do I torture myself with this stuff still...

But after I posted this I did use the precompiled in the Arduino IDE2.0 and it seemed to work. Now will try with VScode and see where I can get.

Thanks, sanity checks are helpful.

pablogs9, 2023-08-16:
> Which I have the other packages as well. Man, I am retired why do I torture myself with this stuff still...

😂😂

> I have ROS running on a VM. But I don't understand I was literally following the instructions linked to the build on running with vscode. So a little confused if it is not meant for Windows/VScode why is it linking me to install VS code and IDE without redirecting me to the other githubs.

It is kinda assumed in this repo, but most of the micro-ROS tools are Linux-only due to the build system tooling required for cross-compiling the library. It should be possible to port this to Win, but we as open-source maintainers do not have enought bandwidth.

> But after I posted this I did use the precompiled in the Arduino IDE2.0 and it seemed to work. Now will try with VScode and see where I can get.

IMO Arduino is the way to go for micro-ROS when you need to make some quick testing, after having the PoC working, you can move to another package. Nice to hear that it works.