micro-ROS/micro_ros_platformio #114: Feature request Wifi transport for RP2040 W

samuk, 2023-08-13:
## Issue template

- Hardware description: rp2040 pico W
- Installation type: Platformio

#### Steps to reproduce the issue
I would like to use Wifi transport with a PicoW

#### Additional information

Is this a hard problem to solve? Or is it just that no one attempted it yet?

pablogs9, 2023-08-14:
Hello @samuk, it should be easy to implement following the structure of any networked transport such as [this one](https://github.com/micro-ROS/micro_ros_platformio/blob/main/platform_code/arduino/native_ethernet/micro_ros_transport.cpp).

It is not in our roadmap, but contributions and PR are welcome.

samuk, 2023-08-14:
Thanks, wouldn't I be using the existing https://github.com/micro-ROS/micro_ros_platformio/blob/main/platform_code/arduino/wifi/micro_ros_transport.cpp

samuk, 2023-08-17:
Note to self. Maybe just drop Platformio & use micro-ros-arduino instead: https://github.com/micro-ROS/micro_ros_arduino/pull/1425

samuk, 2023-09-08:
So it seems the task would be to get Microros running under https://github.com/earlephilhower/arduino-pico

brianlmerritt, 2023-11-24:
@samuk - I haven't had a chance to try it yet with Pico W, but I understand that the pull request https://github.com/platformio/platform-raspberrypi/pull/36 @maxgerhardt has done all the heavy lifting to pull in arduino-pico.  On your VSCode terminal, type `pio pkg install -g -p https://github.com/maxgerhardt/platform-raspberrypi` and that re-adds the platform-raspberrypi with all the earlephilhower goodness.  Platformio haven't accepted the pull request due to licensing issues.

samuk, 2023-11-24:
Thanks, unfortunately MicroROS doesn't run under Earlephilhower.. So that's the remaining task to get that running.

brianlmerritt, 2023-11-24:
It runs under platform.io if you do the above command - I tried it

I just haven't had a chance to test Pico W with it with micro-ros

samuk, 2023-11-24:
Yes I can install and run Earlephillhower and use it under platformio to flash the Pico. I can't get MicroROS to compile in that environment though.

samuk, 2023-11-24:
Or maybe it was just the wifi transport I couldn't enable, it was a while ago that I tried.

brianlmerritt, 2023-11-24:
Arduino RP2040 Connect is supported, and I have one, but am away for a while.  Will have a go hopefully next week.