micro-ROS/micro_ros_platformio #115: rosidl_runtime_cppConfig.cmake missing error

Russ76, 2023-08-24:
## Issue template

- Hardware description: 
- laptop Ryzen7 with 16Gig RAM, Ubuntu Jammy
- RTOS: 
- VScode with Platformio
- Installation type: 
- Attempting to install publisher example script
- Version or commit hash: 
- Humble

#### Steps to reproduce the issue

Followed instructions for installing VScode, Platformio module, MicroROS library

#### Expected behavior
That it would compile example publisher program

#### Actual behavior
Downloads many modules and begins compilation but ends with these errors

#### Additional information
Will include the complete list of warnings and errors

 *  Executing task: platformio run 

Processing teensy40 (platform: teensy; board: teensy40; framework: arduino)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.18.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.158.0 (1.58) 
 - tool-teensy @ 1.158.0 (1.58) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/russ/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/russ/.platformio/penv/lib/python3.10/site-packages (6.0.1)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/russ/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/russ/.platformio/penv/lib/python3.10/site-packages (2.0.1)
Configuring teensy40 with transport serial
Downloading micro-ROS dev dependencies
         - Downloaded ament_cmake
         - Downloaded ament_lint
         - Downloaded ament_package
         - Downloaded googletest
         - Downloaded ament_cmake_ros
         - Downloaded ament_index
Building micro-ROS dev dependencies
Downloading micro-ROS library
         - Downloaded microcdr
         - Downloaded microxrcedds_client
         - Downloaded rcl
         - Downloaded rcl_yaml_param_parser (ignored)
         - Downloaded rcl_action
         - Downloaded rcl_lifecycle
         - Downloaded rclc_lifecycle
         - Downloaded rclc
         - Downloaded rclc_examples (ignored)
         - Downloaded rclc_parameter
         - Downloaded micro_ros_utilities
         - Downloaded rcutils
         - Downloaded micro_ros_msgs
         - Downloaded rmw_microxrcedds
         - Downloaded rosidl_typesupport_c
[... 255 lines cut ...]
  "rosidl_runtime_cpp", but CMake did not find one.

  Could not find a package configuration file provided by
  "rosidl_runtime_cpp" with any of the following names:

    rosidl_runtime_cppConfig.cmake
    rosidl_runtime_cpp-config.cmake

  Add the installation prefix of "rosidl_runtime_cpp" to CMAKE_PREFIX_PATH or
  set "rosidl_runtime_cpp_DIR" to a directory containing one of the above
  files.  If "rosidl_runtime_cpp" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/russ/microros_ws/install/rosidl_typesupport_microxrcedds_cpp/share/rosidl_typesupport_microxrcedds_cpp/cmake/rosidl_typesupport_microxrcedds_cppConfig.cmake:41 (include)
  /home/russ/Documents/PlatformIO/Projects/pub2/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/russ/Documents/PlatformIO/Projects/pub2/.pio/libdeps/teensy40/micro_ros_platformio/build/mcu/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Failed   <<< builtin_interfaces [0.69s, exited with code 1]

=========================== [FAILED] Took 163.73 seconds ===========================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

pablogs9, 2023-09-04:
Do you have your ROS 2 environment source when trying to build this package? Make sure that **it is not sourced**, also check your bashrc to see if it is being sourced there

amalshaji4540, 2023-12-21:
@pablogs9 
I had also faced the same issue. Your solution helped me. **Thank you.**

First I had remove the ROS2 environment source, still I had faced the error. Then after restarting the computer everything is working as expected..