micro-ROS/micro_ros_platformio #119: ament_cmake clone failed when compiling for esp32-dev

kevinEberson, 2023-09-26:
## Issue when cloning micro-ros repo

- Hardware description: ESP32Dev
- Installation type: PlatformIO, Ubuntu Linux 22.04 latest version, git version: 2.34.1, cmake version: 3.22.1, python3-pip version: 22.0.2
- Version or commit hash: Humble

#### Steps to reproduce the issue

- Create a new PlatformIO project using the ESP32-dev as board of choice and add micro-ros dependency to platformio.ini file:

`lib_deps = https://github.com/micro-ROS/micro_ros_platformio`

- Compile the project as normal

#### Expected behavior
Succesfully compile the package

#### Actual behavior
Cloning of micro-ros repo fails:

`
Configuring esp32dev with transport serial
Downloading micro-ROS dev dependencies
ament_cmake clone failed: 
fatal: Too many arguments.

usage: git clone [<options>] [--] <repo> [<dir>]

    -v, --verbose         be more verbose
    -q, --quiet           be more quiet
    --progress            force progress reporting
    --reject-shallow      don't clone shallow repository
    -n, --no-checkout     don't create a checkout
    --bare                create a bare repository
    --mirror              create a mirror repository (implies bare)
    -l, --local           to clone from a local repository
    --no-hardlinks        don't use local hardlinks, always copy
    -s, --shared          setup as shared repository
    --recurse-submodules[=<pathspec>]
                          initialize submodules in the clone
    --recursive ...       alias of --recurse-submodules
    -j, --jobs <n>        number of submodules cloned in parallel
    --template <template-directory>
                          directory from which templates will be used
    --reference <repo>    reference repository
    --reference-if-able <repo>
                          reference repository
    --dissociate          use --reference only while cloning
    -o, --origin <name>   use <name> instead of 'origin' to track upstream
    -b, --branch <branch>
                          checkout <branch> instead of the remote's HEAD
    -u, --upload-pack <path>
                          path to git-upload-pack on the remote
    --depth <depth>       create a shallow clone of that depth
    --shallow-since <time>
                          create a shallow clone since a specific time
    --shallow-exclude <revision>
                          deepen history of shallow clone, excluding rev
    --single-branch       clone only one branch, HEAD or --branch
    --no-tags             don't clone any tags, and make later fetches not to follow them
    --shallow-submodules  any cloned submodules will be shallow
    --separate-git-dir <gitdir>
                          separate git dir from working tree
    -c, --config <key=value>
                          set config inside the new repository
    --server-option <server-specific>
                          option to transmit
    -4, --ipv4            use IPv4 addresses only
    -6, --ipv6            use IPv6 addresses only
    --filter <args>       object filtering
    --remote-submodules   any cloned submodules will use their remote-tracking branch
    --sparse              initialize sparse-checkout file to include only files at root
`

#### Additional information
Same issue as the following issues: #60, #111 

The following platformio.ini configuration is used:

`
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_microros_distro = humble
board_microros_transport = serial
monitor_speed = 115200
lib_deps = 
	adafruit/Adafruit Unified Sensor@^1.1.6
	adafruit/Adafruit BNO055@^1.6.1
	adafruit/Adafruit BusIO@^1.14.1
	SPI
	adafruit/Adafruit NeoPixel@^1.11.0
	https://github.com/micro-ROS/micro_ros_platformio
`

pablogs9, 2023-09-26:
Hello @kevinEberson are you using Plaform.IO in a Windows environment?

kevinEberson, 2023-09-26:
Hi @pablogs9, I am running platfromIO on Ubuntu Linux 22.04 LTS.

kevinEberson, 2023-09-26:
For testing purposes, I commented all code, and copied the implementation of the platformio.ini and main.c file to the project from the micro-ros example. In theory, everything should be the same but I still get the error as mentioned.

pablogs9, 2023-09-26:
Could you add a `print(command)` after this line: 

https://github.com/micro-ROS/micro_ros_platformio/blob/841dc1396eb1d54d61b66781768359ed00d5de6c/microros_utils/repositories.py#L30

To check which is the git command?

kevinEberson, 2023-09-26:
the resulting git command output is as following:
```bash
git clone -b humble https://github.com/ament/ament_cmake /media/kevin/Git drive/git/autonomous-guided-vehicle-platform-3/RIGS-Core/RIGS-Motorcontrol/.pio/libdeps/esp32dev/micro_ros_platformio/build/dev/src/ament_cmake
```
prehaps the error is caused by the space in the path? I will try to validate myself.

pablogs9, 2023-09-26:
Yes, it seems that this space is not smelling well.

Could you check if adding a couple of quotes solves the problem? If it does, could you open a PR with the solution? It would be highly appreciated.

kevinEberson, 2023-09-26:
Formatting folder and self.name as a str() solves the issue in my case. The formatting of the filepath seems to be the issue here. I will look into fixing this in a proper way and create a PR to fix the issue.

pablogs9, 2023-09-26:
Cool, thanks for contributing!

kevinEberson, 2023-09-27:
I tried a couple of things, but I keep running into some issues with either the file path or shutil not recognizing the file path. My expertise is more in lower-level languages like C. I changed the "__init__" function of the Build class to the following implementation:

MojitoShader, 2024-07-11:
@kevinEberson 
Where/how do you open up the Build class? (How can I modify that stuff) I am running into this problem for ever now and I have no Idea how to fix it. It occurs on Windows and with WSL soooo. 
Thx for the help, cheers

hippo5329, 2024-07-11:
Windows/WSL will not build. You should use docker or virtualbox to run ubuntu 22.04 and humble. If you have pi4/5, you may vscode/ssh to edit on windows and compile on Pi.

hippo5329, 2024-07-11:
https://docs.ros.org/en/humble/How-To-Guides/Run-2-nodes-in-single-or-separate-docker-containers.html

MojitoShader, 2024-07-12:
Thanks for the help, I'll try that next.
Is this just a ESP32 Problem or generally mircoRos? 
So the solution is to use the ArduinoIDE 🤢 (wich is able to compile the Example) or VS Code on Ubuntu.
I tried to use the Arduino library as zip in PlatformIO but that didn't work either. Any Idea why that doesn't work? 
I think that would be a nice workaround (Normally Arduino libs are mostly compatible with PIO)

hippo5329, 2024-07-12:
This issue is not specific to esp32. Though it is possible to run ROS2 Jazzy on Windows, it is difficult for beginners. Docker is popular in ROS development.

Platformio and Arduino have minor different micro-ROS implementation. So it won't work to just insert the Arduino library.

VScode is a much powerful IDE than Arduino IDE. If you have pi4/5, it would be better to run vscode on Windows and ssh to Pi. Check my wiki, 

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#development-on-windows-or-linux-desktop-and-use-vscode-remote-development-ssh-to-rpi45

hippo5329, 2024-07-12:
The micro_ros_platformio uses ardunio framework. You can use most of the arduino drivers and libraries. The arduino libraries will be built from source. So it will more flexible than the pre-built arduino library.

hippo5329, 2024-07-12:
WSL might work. Though I do not use Windows and never try out.

https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Webots/Installation-Windows.html

https://code.visualstudio.com/docs/remote/wsl

MojitoShader, 2024-07-13:
It's coming together now. 
The learning curve is pretty steep. But at least I can compile and do stuff now. Thanks for the help. @hippo5329 
I think it would be great to have your Example next to the ESP32 / have examples next to the supported Hardware.
(Here: https://github.com/micro-ROS/micro_ros_platformio) (Or have a examples sub Site)
It so easy to get lost in all the possibilities and versions of ROS / micro_ros.
(I don't know if that makes sense in a git-project manner..)
I am working on a simple esp32 robot (SLAM) with a replacement lidar for vacuum cleaners from Aliexpress.
(They are available for less than 20$$ )

hippo5329, 2024-07-13:
It is a good idea to use cheap vacuum cleaner as a robot base. I recalled that irobot did it before. If you are going to do slam with esp32, you may check another wiki.

[linorobot2_hardware firmware for mobile robots](https://github.com/hippo5329/linorobot2_hardware/wiki)

hippo5329, 2024-07-13:
About the various ROS2 releases, I am moving to the current LTS stable Jazzy. I will let you know when ready. One of the reason is Pi 5, which is supported with Ubuntu 24.04 but not 22.04. Pi 5 is much faster than Pi 4. The price of them are close. Pi 5 should be considered for new purchase.