micro-ROS/micro_ros_platformio #121: MicroROS spin_some() blocks the functionality of the Adafruit bno08x library

snelson-durrant, 2023-10-07:
I'm a college student working on a robotics project that requires interfacing with various sensors on a Teensy 4.1 connected by serial to a Raspberry Pi. Currently, we're using ROS2 Humble on the Pi and microROS on the Teensy to pass messages back and forth:

```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_distro = humble
board_microros_transport = serial
board_microros_user_meta = custom.meta
```

It works great for the majority of our sensors, but we ran into a bug that might be a bit over our heads the other day as we added the Adafruit BNO085 IMU library:

https://github.com/adafruit/Adafruit_BNO08x

We have a state machine in our main loop() on the teensy that manages our agent connection and runs two basic functions: imu_update() (which follows the code of the basic examples on the github extremely closely) and rclc_executor_spin_some():

```
void loop() {
  // FIRST IMPORTANT FUNCTION CALL
  imu_pub.imu_update();

  // state machine to manage connecting and disconnecting the micro-ROS agent
  switch (state) {
  case WAITING_AGENT:
    EXECUTE_EVERY_N_MS(500, state = (RMW_RET_OK == rmw_uros_ping_agent(100, 1))
                                        ? AGENT_AVAILABLE
                                        : WAITING_AGENT;);
    break;

  case AGENT_AVAILABLE:
    state = (true == create_entities()) ? AGENT_CONNECTED : WAITING_AGENT;
    if (state == WAITING_AGENT) {
      destroy_entities();
    };
    break;

  case AGENT_CONNECTED:
    EXECUTE_EVERY_N_MS(200, state = (RMW_RET_OK == rmw_uros_ping_agent(100, 1))
                                        ? AGENT_CONNECTED
                                        : AGENT_DISCONNECTED;);
    if (state == AGENT_CONNECTED) {
      // SECOND IMPORTANT FUNCTION CALL
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
    }
    break;

  case AGENT_DISCONNECTED:
    destroy_entities();
    state = WAITING_AGENT;
    break;

  default:
    break;
  }
}
```

The IMU functionality is a bit different than what I'm used to -- from the library it looks like a "SH-2 Sensor Hub" holds a queue of received data from the IMU (updated on some sort of interrupt callback) that we can check and then pull from in our imu_update function. It works great as long as we're not calling the rclc_executor_spin_some() function -- then the data in the queue disappears until we're not calling spin_some() anymore, whether by disconnecting the agent or simply commenting out the spin_some() line. We've done some pretty extensive testing, and both the IMU update function and the microROS aspects of the code work, just not at the same time.

Is it possible that microROS or the agent is "blocking" the IMU library from executing the internal callbacks that put data in the queue? If so, what would be the best approach to making them work together?

pablogs9, 2023-10-09:
Hello @snelson-durrant, I'm not sure about the usage of your sensor and library. 

How is the library communicating with the sensor? Is it possible that it is interfering with the UART port that the micro-ROS Client is using to communicate with the micro-ROS Agent?

In the case you need to call `imu_pub.imu_update()` at a specific rate, you need to be careful with the micro-ROS part.
Is the timing of `imu_pub.imu_update()` critical for the correct behavior of the sensor library?

snelson-durrant, 2023-10-09:
We're using I2C and tested some varying delays in main() to try and replicate the issue with no success, but it may be an timing issue internal to the library. For now, we've hacked together a couple custom library functions that modify it a bit to get around the parts that don't work when microROS is running, but we'll look into using an RTOS in the future. Thanks for your help!

closed 2023-10-09 by snelson-durrant