micro-ROS/micro_ros_platformio #122: Crashing when connected to Agent over UDP

patrickwasp, 2023-12-04:
## Issue template

- Hardware description: ESP32-POE-ISO
- RTOS: Arduino (FreeRTOS)
- Installation type: platformIO
- Version or commit hash: #8103c3c2bbf52bc2cae5ff95fc61f06dd5792b62 (humble)

#### Steps to reproduce the issue
I made an example repo with Serial and UDP versions of the same basic node, publishing the milliseconds after boot.

The serial version works as expected, but the UDP version reboots every few seconds. The UDP still publishes correctly while it's running.

https://github.com/patrickwasp/esp32-ros-examples

In my `platform.ini` I set `board_microros_transport = custom` and I initialize the connection using 

```
set_microros_ethernet_transports(kMicroControllerIp, kSubnetMask, kAgentIp,
                                   kAgentPort);
```

When I don't have the agent running the micro controller doesn't crash.

The return code error I get is `101`. Is there anything wrong with my setup? 

#### Expected behavior
No restarts

#### Actual behavior
Restarts after a few seconds

#### Additional information

Serial output
```
Ethernet Connected, MAC: 40:22:D8:17:99:2F
Setup complete
5594
6441
6945
7441
8014
8445
ROS error: 101
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
```

pablogs9, 2023-12-05:
Hello, if you are working with ESP32, the recommended and tested platform is [micro-ROS component for ESP-IDF
](https://github.com/micro-ROS/micro_ros_espidf_component). Could you test that the basic example there (an UDP example) works properly?

patrickwasp, 2023-12-05:
Okay, I'll have to figure out how to use IDF to test that out. Any tips on how to set up IDF with ROS on platformIO?

In the meantime, I have added more debugging output so it looks like the agent regularly disconnects and sometimes fails to reinitialize. This means there are two problems, 1,  it is disconnecting, and 2, it cannot reliably reconnect.

This is the function I call when the agent is available, and it fails the check on `rclc_node_init_default` with the error RCL_RET_NOT_INIT. Are there any options to make that call more robust? Would putting that line in a while loop to wait until it does connect be a good solution? 

```
bool CreateEntities() {
  g_ros_allocator = rcl_get_default_allocator();

  g_ros_init_options = rcl_get_zero_initialized_init_options();
  RC_CHECK(rcl_init_options_init(&g_ros_init_options, g_ros_allocator));
  RC_CHECK(rcl_init_options_set_domain_id(&g_ros_init_options, kDomainId));
  rclc_support_init_with_options(&g_ros_support, 0, NULL, &g_ros_init_options,
                                 &g_ros_allocator);

  RC_CHECK(rclc_node_init_default(&g_ros_node, kNodeName, kNamespace,
                                  &g_ros_support));

  g_ros_executor = rclc_executor_get_zero_initialized_executor();
  RC_CHECK(rclc_executor_init(&g_ros_executor, &g_ros_support.context,
                              kNumberOfHandles, &g_ros_allocator));

  InitializePublishersAndTimers();

  UpdateTimeOffsetFromAgent();
  return true;
}
```

patrickwasp, 2023-12-05:
I may have found a work-around for the disconnecting problem, but I not the crashing.

I changed this part of my code while checking for connectivity to the agent:

```cpp
rmw_uros_ping_agent(100, 1)
```
to
```cpp
rmw_uros_ping_agent(200, 3)
```

It used to disconnect every 2-5 seconds, now it's connected for the past hour. I'll let it run for a few more hours to see if this solves problem 1.

closed 2024-03-21 by patrickwasp