micro-ROS/micro_ros_platformio #124: Windows 11 VSCode "'.' is not recognized as an internal or external command" when adding library to lib_deps in platformio.ini

Darkextratoasty, 2023-12-19:
## Issue template

- Hardware description: 
Windows 11 PC running Visual Studio Code 1.85.1 with PlatformIO 6.1.11 extension.

- Installation type: 
Adding the lines:
```
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```
to the platformio.ini file.

- Version or commit hash: 
Unsure how to check this, I got the URL from the "https://github.com/micro-ROS/micro_ros_platformio" webpage using the Code > Clone > HTTPS button on 12/19/2027. It appears that the default is iron, but I don't know how to confirm that.

#### Steps to reproduce the issue

- In VSCode go to PlatformIO extension > Quick Access > PIO Home > Open > + New Project
- Select Teensy 4.1 (or any board, the issue doesn't seem to be tied to the board)
- Once the project is initialized, paste the config from here: https://github.com/micro-ROS/micro_ros_platformio/blob/main/examples/micro-ros_publisher/platformio.ini into the platformio.ini file
```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
board_microros_transport = serial
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```
- save file

*Note:* this procedure works fine on Ubuntu 22.04.

#### Expected behavior
Project should configure properly.

#### Actual behavior
Get the error:
```
Build dev micro-ROS environment failed: 
 '.' is not recognized as an internal or external command,
```
#### Additional information
The full output is below:

```
Resolving teensy41 dependencies...
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio
git version 2.39.0.windows.2
Cloning into 'C:\Users\joshu\.platformio\.cache\tmp\pkg-installing-atkqsg_i'...

Library Manager: micro_ros_platformio@0.0.1+sha.e87705f has been installed!
Updating metadata for the vscode IDE...
UserSideException: Processing teensy41 (platform: teensy; board: teensy41; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.18.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.158.0 (1.58) 
 - tool-teensy @ 1.158.0 (1.58) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
C:\Users\joshu\.platformio\penv\Scripts\python.exe -m pip install pyyaml
Requirement already satisfied: pyyaml in c:\users\joshu\.platformio\penv\lib\site-packages (6.0.1)

[notice] A new release of pip available: 22.2.2 -> 23.3.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Installing markupsafe==2.0.1 with pip at PlatformIO environment
C:\Users\joshu\.platformio\penv\Scripts\python.exe -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in c:\users\joshu\.platformio\penv\lib\site-packages (2.0.1)

[notice] A new release of pip available: 22.2.2 -> 23.3.2
[notice] To update, run: python.exe -m pip install --upgrade pip
Configuring teensy41 with transport serial
Downloading micro-ROS dev dependencies
	 - Downloaded ament_cmake
	 - Downloaded ament_lint
	 - Downloaded ament_package
	 - Downloaded googletest
	 - Downloaded ament_cmake_ros
	 - Downloaded ament_index
Building micro-ROS dev dependencies
Build dev micro-ROS environment failed: 
 '.' is not recognized as an internal or external command,

operable program or batch file.

========================== [FAILED] Took 8.88 seconds ==========================
```

joshua-8, 2024-01-02:
Hi, I get the same issue on Windows 10.

The . comes from https://github.com/micro-ROS/micro_ros_platformio/blob/e87705fab34ff23f2d117d0185f19fc8698ff081/microros_utils/library_builder.py#L103C29-L103C31
where it appears that it's being used as a synonym for the source command: https://ss64.com/bash/source.html

I don't know why neither "source" nor "." are recognized in the environment that the commands run in. 

Hopefully we can keep each other updated as we troubleshoot this and maybe we'll find a solution.

joshua-8, 2024-01-02:
I just found another potential issue:
In my .platformio folder (in my windows user folder), there's a folder named penv, but within that there is no /bin folder, only a Scripts folder, though interestingly there is a file called activate, activate.bat, and Activate.ps1 among other files.
That might be relevant to this line: https://github.com/micro-ROS/micro_ros_platformio/blob/e87705fab34ff23f2d117d0185f19fc8698ff081/extra_script.py#L97

Legohead259, 2024-01-06:
It could be that Micro-ROS is not intended to be compiled on Windows? 

For your application, is there a specific need to use VSCode on Windows or would WSL2 Ubuntu suffice?