micro-ROS/micro_ros_platformio #125: Micro-ROS parameter server not reporting correct parameter values to host on initial connection

Legohead259, 2024-01-06:
## Issue template

- Hardware description: **Host PC:** Windows10 WSL2 w/ Ubuntu 22.04.3 and ROS2, **MCU** ESP32
- RTOS: FreeRTOS on ESP32
- Installation type: PlatformIO and MicroROS module on host PC
- Version or commit hash: Iron

#### Steps to reproduce the issue

1. Flash firmware to ESP32 dev board (https://github.com/Legohead259/MicroROS-ESC-Firmware/tree/parameter-server)
2. Connect ROS2 agent to board
3. Execute `ros2 param list`
5. Execute `ros2 param get /bldc_node [param name]`
6. Execute `ros2 param set /bldc_node [param_name] [param_value]`
7. Execute `ros2 param get /bldc_node [param_name]`
8. Be confused

#### Expected behavior
<details>
<summary>Parameter List Output</summary>

**Works as expected**

```bash
/bldc_node:
  angle_d
  angle_i
  angle_lpf_tf
  angle_p
  angle_ramp
  controller_mode
  driver_v_limit
  kv_rating
  motor_i_limit
  motor_v_limit
  motor_vel_limit
  phase_l
  phase_r
  pole_pairs
  velocity_d
  velocity_i
  velocity_lpf_tf
  velocity_p
  velocity_ramp
```

</details>

Expected output from: `ros2 param get /bldc_node contoller_node`:
    
    Integer value is: 7

Expected output from `ros2 param set /bldc_node controller_mode 5 && ros2 param get /bldc_node controller_mode`
    
    Set parameter successful
    Integer value is: 5
    
#### Actual behavior

Actual output from: `ros2 param get /bldc_node contoller_node` on initial connection:

    Integer value is: 0

But, the external serial debugger reports that the controller mode is successfully set to 7 on Micro-ROS parameter server start up. Additionally, the parameter server functions as expected. Changing the parameter from the host PC successfully changes the parameter value on the Micro ROS client and then the `ros2 param get` command reports the correct value.

#### Additional information

Here is the code where the parameter service is initialized and the parameters added. I'm pretty sure I followed the parameter server tutorial properly from the docs.

<details>
[... 510 lines cut ...]
    motor.P_angle.output_ramp = param->double_value;
}

void angleLFPChangeCallback(parameter_t* param) {
    motor.LPF_angle.Tf = param->double_value;
}

void driverVoltageLimitChangeCallback(parameter_t* param) {
    driver.voltage_limit = param->double_value;
}

void motorVoltageLimitChangeCallback(parameter_t* param) {
    motor.voltage_limit = param->double_value;
}

void motorCurrentLimitChangeCallback(parameter_t* param) {
    motor.current_limit = param->double_value;
}

void motorVelocityLimitChangeCallback(parameter_t* param) {
    motor.voltage_limit = param->double_value;
}
```

</details>

pablogs9, 2024-01-08:
Could you check if a simpler example like this one: https://github.com/micro-ROS/micro_ros_espidf_component/blob/iron/examples/parameters/main/main.c works properly on your platform?

Legohead259, 2024-02-08:
Found the error! I am using a lookup array to keep track of all the parameter objects. In my implementation, each parameter I want to trigger a callback to change a value with the electronic speed controller. For example: changing the controller mode parameter triggers a callback that performs some `controller_mode`-specific logic for the change. Same with the other parameters. To that end, the parameter change callback for the parameter server used to look like this:
```cpp
bool onParameterChangedCallback(const Parameter* oldParam, const Parameter* newParam, void* context) {
    (void) context;

    if (oldParam == NULL && newParam == NULL) { // Check for callback error
        Serial1.printf("Callback error, both parameters are NULL\n"); // Debug
        return false;
    }

    if (oldParam == NULL) { // Check for new parameter
        Serial1.printf("Creating new parameter %s\n", newParam->name.data); // Debug
        return false; // Do not allow new parameters
    } 
    else if (newParam == NULL) { // Check for deleting parameter
        Serial1.printf("Deleting parameter %s\n", oldParam->name.data); // Debug
        return false; // Do not allow deleting parameters
    } 
    else { // Check for changing parameters
        Serial1.printf("Checking for: %s\r\n", newParam->name.data);
        for (uint8_t i=0; i<NUM_PARAMETERS; i++) { // Iterate through the parameter array
            Serial1.printf("\t Comparing against: %s\r\n", params[i]->key);
            if (strcmp(newParam->name.data, params[i]->key) == 0) { // Check if parameter is in array
                Serial1.printf("Editing %s\r\n", params[i]->key); // DEBUG
                switch (newParam->value.type) { // Check parameter type and set new value accordingly
                    case RCLC_PARAMETER_BOOL:
                        params[i]->bool_value = newParam->value.bool_value;
                        break;
                    case RCLC_PARAMETER_INT:
                        params[i]->integer_value = newParam->value.integer_value;
                        break;
                    case RCLC_PARAMETER_DOUBLE:
                        params[i]->double_value = newParam->value.double_value;
                        break;
                    default:
                        break;
                }
                params[i]->onChange(); // Execute the parameters change function
                saveParam(params[i]);
                break;
            }
            else { return false; }
        }
    }

    return true;
}
```
<details>

```
Adding parameters to server...
Checking for: controller_mode
         Comparing against: controller_mode
Editing controller_mode
Set motor control mode Using callback!
Checking for: pole_pairs
         Comparing against: controller_mode
Checking for: phase_r
         Comparing against: controller_mode
Checking for: kv_rating
         Comparing against: controller_mode
Checking for: phase_l
         Comparing against: controller_mode
Checking for: velocity_p
         Comparing against: controller_mode
Checking for: velocity_i
         Comparing against: controller_mode
Checking for: velocity_d
         Comparing against: controller_mode
[... 276 lines cut ...]
         Comparing against: phase_l
         Comparing against: velocity_p
         Comparing against: velocity_i
         Comparing against: velocity_d
         Comparing against: velocity_ramp
         Comparing against: velocity_lpf_tf
         Comparing against: angle_p
         Comparing against: angle_i
         Comparing against: angle_d
         Comparing against: angle_ramp
         Comparing against: angle_lpf_tf
         Comparing against: driver_v_limit
         Comparing against: motor_v_limit
         Comparing against: motor_i_limit
         Comparing against: motor_vel_limit
Editing motor_vel_limit
</details>

Expected and correct output from ROS2 CLI:
```bash
~/ROS2-Micro-ROS-ESC-Workspace$ ros2 param get bldc_node angle_p
Double value is: 20.0
~/ROS2-Micro-ROS-ESC-Workspace$ ros2 param set bldc_node angle_p 21.0
Set parameter successful
```

closed 2024-02-08 by Legohead259