micro-ROS/micro_ros_platformio #129: Publishing to a subscriber causes rclc_executor_spin_some to return a fail code.

AndyBlightLeeds, 2024-03-15:
## Issue template

- Hardware description: ESP32CAM board
- RTOS: FreeRTOS
- Installation type: PlatformIO with the Arduino framework.
- Version or commit hash: rolling 

#### Steps to reproduce the issue

I have created a subscriber example on my fork of this repo that shows the problem.  The URL of the branch is <https://github.com/AndyBlightLeeds/micro_ros_platformio/tree/add-subscriber-example>.  Happy to add this example once the bug is fixed.

1. Clone the repo in the link and checkout the `add-subscriber-example` branch. 
2. Navigate to the PlatformIO project <https://github.com/AndyBlightLeeds/micro_ros_platformio/tree/add-subscriber-example/examples/micro-ros_subscriber>
3. Open `Main.cpp` in your editor and set the `SSID`, `SSID_PW` and `agent_ip` values to suit your system.
4. Build and upload the project using `pio run -t upload`.
5. Start the micro-ROS agent on your PC. 
6. Attach a serial monitor program to the ESP32CAM.
7. Start the ESP32CAM.  You should see the following output:
    ```text
    st:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
    configsip: 0, SPIWP:0xee␍␊
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
    mode:DIO, clock div:2␍␊
    load:0x3fff0030,len:1184␍␊
    load:0x40078000,len:13232␍␊
    load:0x40080400,len:3028␍␊
    entry 0x400805e4␍␊
    Started␍␊
    Connected␍␊
    Running␍␊
    Looping␍␊
   ```  
9. On your PC, open another window, source your ROS installation `source /opt/ros/rolling/setup.bash`.  Verify that the topic `/cmd_vel` is being published. 
10. Send the following command: 
    ```bash
    ros2 topic pub -1 /cmd_vel geometry_msgs/msg/Twist "{linear:{x: 1.0}, angular:{z: -1.0}}"
    ```
11. On the serial monitor, I get the following output:
    ```text
    Looping␍␊
    Received twist message␍␊
    Looping␍␊
    Failed status on line 126: 1. Continuing.␍␊
    Looping␍␊
    Failed status on line 126: 1. Continuing.␍␊
    ```
12. If the twist message is sent again, the text `Received twist message` is not seen again. 

#### Expected behavior

The twist message should be received and the function `rclc_executor_spin_some` should not fail.  Subsequent twist messages should be processed.

#### Actual behavior

The function `rclc_executor_spin_some` fails and no further twist messages are handled.

#### Additional information

The code is based on the publisher example and there are a couple of `#define` values that can be set enable publisher only, subscriber only or both.  The publisher works fine, just the subscriber fails.

I have dealt with something like this before.  It would be great if the result code was more meaningful instead of 0 for OK and 1 for something has gone wrong.  For instance, if there is a resource issue caused by `colcon.meta` not being set correctly, you might return a different number.  It would also be great if the result codes were documented so we know what they mean.

pablogs9, 2024-03-18:
Twist message is a pointer: 

https://github.com/AndyBlightLeeds/micro_ros_platformio/blob/6d93c0658eb830447bd9b1edb7e36a7d917f2d51/examples/micro-ros_subscriber/src/Main.cpp#L25

And it is being referenced in the executor with no initialization: 

https://github.com/AndyBlightLeeds/micro_ros_platformio/blob/6d93c0658eb830447bd9b1edb7e36a7d917f2d51/examples/micro-ros_subscriber/src/Main.cpp#L139C38-L139C47

Example of how to handle correctly messages and pointers:

https://github.com/micro-ROS/micro_ros_arduino/blob/iron/examples/micro-ros_subscriber/micro-ros_subscriber.ino

AndyBlightLeeds, 2024-03-18:
Many thanks for your quick and accurate reply.  I should have spotted it myself!
For future reference, the change was from this: 
```cpp
static geometry_msgs__msg__Twist *twist_msg;`
``` 
to this 
```cpp
static geometry_msgs__msg__Twist twist_msg;
``` 
I'll tidy this up and create a pull request for the new example.

closed 2024-03-18 by AndyBlightLeeds