micro-ROS/micro_ros_platformio #135: Failer build the firmware

phn1712002, 2024-05-23:
- Hardware description: Ubuntu 22.04.4 LTS
- Version or commit hash: ROS2 - Humble

I made a simple tutorial to be able to implement uROS2 on ESP following this tutorial link: [https://technologiehub.at/project-posts/micro-ros-on-esp32-tutorial/](url) and after "pio run" I appeared like this

```
Building micro-ROS library
OSError: [Errno 22] Invalid argument: '/home/ubuntu/uROS2_ws/firmware/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/aux':
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/espressif32/builder/main.py", line 368:
    target_elf = env.BuildProgram()
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProjectDeps()
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 152:
    plb = env.ConfigureProjectLibBuilder()
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 1202:
    project.install_dependencies()
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 987:
    if _is_builtin(spec):
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 979:
    for lb in self.env.GetLibBuilders():
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 1115:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 153:
    self.process_extra_options()
  File "/root/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piolib.py", line 301:
    self.env.SConscript(
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/ubuntu/uROS2_ws/firmware/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/extra_script.py", line 158:
    build_microros()
  File "/home/ubuntu/uROS2_ws/firmware/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/extra_script.py", line 100:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "/home/ubuntu/uROS2_ws/firmware/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/microros_utils/library_builder.py", line 67:
    self.package_mcu_library()
  File "/home/ubuntu/uROS2_ws/firmware/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/microros_utils/library_builder.py", line 191:
    os.makedirs(aux_folder, exist_ok=True)
  File "/usr/lib/python3.10/os.py", line 225:
    mkdir(name, mode)
======================================================== [FAILED] Took 1012.52 seconds ========================================================
```

Latek, 2024-05-24:
I got more or less the same stack trace.

```
> pio run
Processing due (platform: atmelsam; board: due; framework: zephyr)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM (8.0.0) > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: 
 - framework-zephyr @ 2.20701.220422 (2.7.1) 
 - tool-cmake @ 3.21.3 
 - tool-dtc @ 1.4.7 
 - tool-gperf @ 3.1.0 
 - tool-ninja @ 1.9.0 
 - toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing importlib-resources with pip at PlatformIO environment
/usr/local/Cellar/platformio/6.1.15/libexec/bin/python -m pip install importlib-resources
Requirement already satisfied: importlib-resources in /usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages (6.4.0)
Installing pyyaml with pip at PlatformIO environment
/usr/local/Cellar/platformio/6.1.15/libexec/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages (6.0.1)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/usr/local/Cellar/platformio/6.1.15/libexec/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages (2.0.1)
Configuring due with transport serial
TypeError: sequence item 0: expected str instance, tuple found:
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "~/.platformio/platforms/atmelsam/builder/main.py", line 127:
    target_elf = env.BuildProgram()
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "~/.platformio/packages/framework-zephyr/scripts/platformio/platformio-build-pre.py", line 26:
    env.ProcessProjectDeps()
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 152:
    plb = env.ConfigureProjectLibBuilder()
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 1202:
    project.install_dependencies()
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 987:
    if _is_builtin(spec):
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 979:
    for lb in self.env.GetLibBuilders():
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 1115:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 153:
    self.process_extra_options()
  File "/usr/local/Cellar/platformio/6.1.15/libexec/lib/python3.12/site-packages/platformio/builder/tools/piolib.py", line 301:
    self.env.SConscript(
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "~/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "~/dev/iot/ros2-due-demo/.pio/libdeps/due/micro_ros_platformio/extra_script.py", line 157:
    build_microros()
  File "~/dev/iot/ros2-due-demo/.pio/libdeps/due/micro_ros_platformio/extra_script.py", line 93:
    "{} {} -DCLOCK_MONOTONIC=0 -D'__attribute__(x)='".format(' '.join(env['CFLAGS']), ' '.join(env['CCFLAGS'])),
```

Here is my system information:
```
pio system info
--------------------------  ------------------------------------------------------
PlatformIO Core             6.1.15
Python                      3.12.3-final.0
System Type                 darwin_x86_64
Platform                    macOS-14.4.1
File System Encoding        utf-8
Locale Encoding             UTF-8
(...)
Python Executable           /usr/local/Cellar/platformio/6.1.15/libexec/bin/python
Global Libraries            0
Development Platforms       4
Tools & Toolchains          32
--------------------------  ------------------------------------------------------
```

hippo5329, 2024-05-28:
@Latek framework = arduino

hippo5329, 2024-05-29:
@phn1712002 I followed the tutorial. There is no such error. If you want to build micro-ROS robot. You may follow my wiki https://github.com/hippo5329/linorobot2_hardware/wiki 

Building micro-ROS library
Found 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.48096f8
Building in release mode
Compiling .pio/build/esp32doit-devkit-v1/src/main.cpp.o
Building .pio/build/esp32doit-devkit-v1/bootloader.bin
Generating partitions .pio/build/esp32doit-devkit-v1/partitions.bin
esptool.py v4.7.3
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
...
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32doit-devkit-v1/libFrameworkArduino.a
Indexing .pio/build/esp32doit-devkit-v1/libFrameworkArduino.a
Linking .pio/build/esp32doit-devkit-v1/firmware.elf
Retrieving maximum program size .pio/build/esp32doit-devkit-v1/firmware.elf
Checking size .pio/build/esp32doit-devkit-v1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.6% (used 34600 bytes from 327680 bytes)
Flash: [==        ]  15.6% (used 204605 bytes from 1310720 bytes)
Building .pio/build/esp32doit-devkit-v1/firmware.bin
esptool.py v4.7.3
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
======================== [SUCCESS] Took 147.52 seconds ========================

phn1712002, 2024-05-29:
@hippo5329, Thank you, but I found the problem, it seems I use windows and ubuntu in parallel, I save my code on a removable hard drive, when the code runs build and initializes the aux folder, the system cannot initialize it. 

```
OSError: [Errno 22] Invalid argument: '/home/ubuntu/uROS2_ws/firmware/.pio/libdeps/esp32doit-devkit-v1/micro_ros_platformio/build/aux':
```

I changed library_builder.py in folder libdeps/esp32doit-devkit-v1/micro_ros_platformio/microros_utils at location def package_mcu_library "/aux" -> "/aux_fix"
```
def package_mcu_library(self):
        binutils_path = self.resolve_binutils_path()
        //aux_folder = self.build_folder + "/aux" 
        aux_folder = self.build_folder + "/aux_fix" 

        shutil.rmtree(aux_folder, ignore_errors=True)
        shutil.rmtree(self.library_path, ignore_errors=True)
        os.makedirs(aux_folder, exist_ok=True)
        os.makedirs(self.library_path, exist_ok=True)
        for root, dirs, files in os.walk(self.mcu_folder + "/install/lib"):
            for f in files:
                if f.endswith('.a'):
                    os.makedirs(aux_folder + "/naming", exist_ok=True)
                    os.chdir(aux_folder + "/naming")
                    os.system("{}ar x {}".format(binutils_path, root + "/" + f))
                    for obj in [x for x in os.listdir() if x.endswith('obj')]:
                        os.rename(obj, '../' + f.split('.')[0] + "__" + obj)

        os.chdir(aux_folder)
        command = "{binutils}ar rc libmicroros.a $(ls *.o *.obj 2> /dev/null); rm *.o *.obj 2> /dev/null; {binutils}ranlib libmicroros.a".format(binutils=binutils_path)
        result = run_cmd(command)

        if 0 != result.returncode:
            print("micro-ROS static library build failed: \n{}".format(result.stderr.decode("utf-8")))
            sys.exit(1)

        os.rename('libmicroros.a', self.library)

        # Copy includes
        shutil.copytree(self.build_folder + "/mcu/install/include", self.includes)

        # Fix include paths
        include_folders = os.listdir(self.includes)

        for folder in include_folders:
            folder_path = self.includes + "/{}".format(folder)
            repeated_path = folder_path + "/{}".format(folder)

            if os.path.exists(repeated_path):
                shutil.copytree(repeated_path, folder_path, copy_function=shutil.move, dirs_exist_ok=True)
                shutil.rmtree(repeated_path)
```

And I was finally able to build it successfully.

Latek, 2024-06-20:
> @Latek framework = arduino

That is correct!