micro-ROS/micro_ros_platformio #143: tf2_msgs not in micro_ros_platformio

kjrusscher, 2024-07-11:
- Hardware description: Teensy 4.1
- RTOS: Arduino
- Installation type: micro_ros_platformio 2.0.0
- Version or commit hash: humble

It's as simple as the title says; tf2_messages are not found in my micro_ros_platformio installation. It is in the [micro_ros_arduino source code](https://github.com/micro-ROS/micro_ros_arduino/tree/humble/src). I could maybe add it via <Project_directory>/extra_packages/extra_packages.repos, but this is not desirable solution for us.

My question: Am I doing something wrong or is this a issue that can be fixed (how?)?

hippo5329, 2024-07-11:
Tf should be handled on robot computer with robot_state_publisher or joint_state_publisher etc. Why do you want tf on micro-controller?

micro_ros_platformio does not include all the msgs that the micro_ros_arduino has. You will need to add them with extra_packages.

kjrusscher, 2024-07-11:
I want to publish the static tf of the robot setup. I have several robots with varying layouts. The robot computer can be switched between the robots. 

I've solved it now by publishing a stamped transform with reliable reliability QoS.

closed 2024-07-15 by kjrusscher