micro-ROS/micro_ros_platformio #149: Localhost configuration with MicroROS and CycloneDDS

ValerioSpagnoli, 2024-09-03:
I'm working with CycloneDDS for ROS2 Humble and I'm trying to use the ROS_LOCALHOST_ONLY variable (or another localhost configuration) with MicroROS. While it worked with FastDDS, I haven't been able to establish communication between the agent and the ROS system using CycloneDDS, even though both are running on the same machine.

The (serial) connection between the agent and the microcontroller is successfully established, and all instances (subscribers, publishers, etc.) are created without issues. However, the topics are not visible in any other terminal on the machine.

Everything run on Ubuntu22/Jetpack6.

Is there an option to enable this configuration or will it be included in the future?

hippo5329, 2024-09-03:
In my experience, micro-ROS does not like Cyclonedds, especially Jazzy. But you may try this,

https://cyclonedds.io/docs/cyclonedds/latest/config/index.html

export CYCLONEDDS_URI="<CycloneDDS><Domain><General><NetworkInterfaceAddress>127.0.0.1</NetworkInterfaceAddress></General></Domain></CycloneDDS>"

ValerioSpagnoli, 2024-09-04:
Thank you, I tried your configuration modified as follow to make it working on Humble, but I got the same problem :(

```
<General>
   <Interfaces>
      <NetworkInterface address="127.0.0.1"> </NetworkInterface>
   </Interfaces>
</General>
```